spacetelescope/mirage #10: Update to use position-dependent PSFs

bhilbert4, 2017-11-16:
Currently PSFs are constant across a detector and in fact the same for all detectors. New webbpsf includes field dependence.

Also need to update to more realistic wavefront error values. Current PSF library is too conservative in terms of WFE.

bhilbert4, 2017-11-17:
This might take a little time. It sounds like Marshall has someone working on an interpolator function, which would be potentially useful. PSF dependence is based on observed PSFs at some irregular grid of locations on the detector. When a user asks for a PSF at location x,y, webbpsf currently returns the PSF at the nearest grid point. 

So for the simulator, I can either create PSFs at these grid points and then write my own interpolator, or wait for the webbpsf interpolator. 

I'm pushing on by creating an updated PSF library with no field dependence (except for detector-dependence), using the requirements and predicted WFE values.

Matteo looked at PSF differences across a detector and said that differences (in? photometry?) were less than 2% across the FOV, so it seems like this is not a large effect.

bhilbert4, 2017-11-29:
Updated PSF library without field dependence (but with detector dependence) is complete. Code has been updated to deal with this new library. #14

laurenmarietta, 2018-08-07:
Being implemented in #117

closed 2018-08-07 by laurenmarietta