spacetelescope/mirage #17: Documentation needed

bhilbert4, 2017-11-29:
Need to write a technical report in order to fully describe the simulator (also be sure to include ancillary useful code, like the yaml_generator). 

TR is in progress. Needs lots of updates from when I started writing it months ago to describe a much less mature code base.

bhilbert4, 2018-01-23:
TR is essentially done. Waiting for IT to get back to me on where to host the repository of config/reference files.

bhilbert4, 2018-03-06:
TR submitted to SOCCER. Comments from independent review should come back relatively soon.

Also to do: proper (PEP8, PEP256) documentation for each function.

bhilbert4, 2018-08-15:
I think example notebooks are the best place to start. I'll also send you a copy of the "handbook" which is in TR form in the middle of the SOCCER review process at the moment.

laurenmarietta, 2018-08-16:
When I was teaching myself how MIRAGE worked, I tried to develop a kind of "outline" that lists everything that happens when the simulator runs. 

Attaching it here; maybe it (or some variation of it) could be useful to include in the documentation somewhere:
[RampSimulatorOutline.docx](https://github.com/spacetelescope/mirage/files/2294062/RampSimulatorOutline.docx)

bhilbert4, 2018-08-16:
That's true. I need to learn Sphinx so that `mirage` can have better documentation. The NIRCam team has also been pushing for me to turn the TR into a PASP paper in order to make it more accessible (and to make a better advertisement).

mperrin, 2018-08-16:
Sphinx is super easy. I'd be happy to set up the basic framework in a PR, and then you can paste in content into that over time.

mperrin, 2018-08-16:
See #147