spacetelescope/mirage #50: Tangent plane projection

bhilbert4, 2018-03-07:
Investigate whether the source locations are correct, or if a tangent plane projection adjustment is needed. 

@Johannes-Sahlmann already has a function that makes this adjustment.

Johannes-Sahlmann, 2018-05-17:
This will be required for enabling high-fidelity astrometric simulations. Attached here are the residuals of a NIRISS simulation of the LMC field after fitting a fourth-degree polynomial. I suspect that the ring structure originates in the missing tangent-plane-projection.

[V1087001001P0000000001101_NIS_CLEAR_linear_dms_rate_residuals_sky.pdf](https://github.com/spacetelescope/nircam_simulator/files/2013138/V1087001001P0000000001101_NIS_CLEAR_linear_dms_rate_residuals_sky.pdf)

Johannes-Sahlmann, 2018-10-01:
@bhilbert4 Was the tangent plane projection implemented already?
I repeated the simulation and analysis with the latest mirage version and do not find the ring structure in the residuals anymore.
In any case, can you please point me to the place in the code where this should be introduced. Functions for tangent plane projection are included in pysiaf, we I will use that dependency.

bhilbert4, 2018-10-01:
No, tangent plane projection has not been implemented. Maybe the residuals have changed due to the switch to pysiaf?

My first thought is that tangent plane calculations should be introduced within the `XYToRADec` and `RADecToXY_astrometric` functions.

Johannes-Sahlmann, 2018-10-11:
@KevinVolkSTScI : tangent plane projection was not yet implemented but the gross errors I had seen previously seem to have disappeared (see this thread). 
I will implement it when I have the proper setup to confirm its validity, i.e. after the NIRISS+FGS parallels are finished.

Johannes-Sahlmann, 2019-04-30:
After re-consideration I state that any projection effects should be absorbed by the distortion coefficients. Hence, this issue can be closed and obsolete.

closed 2019-04-30 by Johannes-Sahlmann