spacetelescope/mirage #62: Add support for FGS imaging

Johannes-Sahlmann, 2018-03-26:
This simulator should be expanded to support FGS imaging as well (in addition to the recently added NIRISS capability).

The main body of work is probably to assemble all the necessary input files and the PSF library, although Kevin Volk has done most of that work already.

bhilbert4, 2018-03-26:
I added the basic information to enable FGS while implementing NIRISS imaging. Once the input files and PSF library for FGS exist, I can test to make sure everything is working.

KevinVolkSTScI, 2018-03-26:
I have copied over a set of PSF files for Guider 1 into 

/ifs/jwst/wit/niriss/nircam_ramp_simulation_files/webbpsf_files/G1/predicted

The names are of the form

fgs_G1_x1024_y1024_predicted_0_m0p50_m0p50.fits

These are the PSF files made with webbpsf 0.6.0rc1 last year, assuming an A0V spectrum.  The PSF depends on the spectral type because of the wide wavelength range of the Guider response and because most of the photons are coming from short wavelengths.  

Note that when I made these files there were 10 realizations of the OTE wavefront error for the Guider case, this I think was when the WebbPSF data files were codeV, but now with codeW there is only one realization.  It may be required to generate new PSF files for the Guider with the current WebbPSF codeW data files to be consistent.

A few of the other required Guider input files have been put in directory

/ifs/jwst/wit/niriss/nircam_ramp_simulation_files/guider_ramp_simulation_files

I am assuming that the guider detector names will be G1 and G2 in the simulation code in the way the PSF files are named.  Is this correct?

KevinVolkSTScI, 2018-03-26:
Sorry I hit the "Close and Comment" button not knowing that it would close the issue.

bhilbert4, 2018-03-26:
Looks like the official DMS detector names for FGS are GUIDER1 and GUIDER2. I can update input files as I go along. I was able to download SIAF for FGS, but am having strange permission problems with the file. It won't let me save as a csv. Once I get that figured out, it sounds like we should be very close to being able to test.

KevinVolkSTScI, 2018-03-26:
I will rename the PSF files to use GUIDER1 and GUIDER2.

bhilbert4, 2018-03-26:
I'm copying over necessary CRDS reference files.

bhilbert4, 2018-03-26:
For the moment, I've copied FGS reference files into: 
/ifs/jwst/wit/nircam/nircam_simulator_data/fgs_reference_files/

Eventually we should put all of the reference files in one place, just to make things easier. This way when the tool is publicly released, it will be easier to create tarballs of the reference files and place them wherever the Archive Branch wants them. 

I've also sent an email to Mike Wolfe, asking if he has DMS-formatted FGS dark exposures that we can use.

KevinVolkSTScI, 2018-03-27:
I have revised the Guider throughput files from what I originally posted in the directory 

/ifs/jwst/wit/niriss/nircam_ramp_simulation_files/guider_ramp_simulation_files

because i realized that the Jansky -> count rate factors listed in the zero magnitude file include the photon yield (as they should...), so the throughput files should also include the photon yield factor to produce the correct result if the code actually were to integrate a spectrum over the response curve to get the mean flux density and the expected count rate.  There are now Guider throughput files with and without the photon yield factor in the directory

guider1_throughput_nopy.txt  guider2_throughput_nopy.txt
guider1_throughput_py.txt    guider2_throughput_py.txt

If one wanted to determine the photon yield value for a source of a known spectral shape then one could integrate the count rate for the _py.txt file and for the _nopy.txt file and take the ratio of the output count rates.  That will give the photon yield factor.  Then in the code one would need carry out the photon by photon Poisson calculation with the count rate per pixel per frame.  All this is possible, but probably it is too much work for the Guider because the photon yield value will vary with the source spectral type.  I can quantify this somewhat to go along with the PSFs that are a function of spectral type later, if Johannes wants this in the code.

For NIRISS it is sufficient to just have the photon yield values for each of the filters.  These values will be somewhat inaccurate for a very cool star or for a dusty object, but in general with the narrower passband the variations should not be that large for most types of objects.

The way the mean flux density for the filters, or for the Guiders, is defined means that the count rate for a given A0V magnitude is correct for any spectral type, but it is not accurate for the AB and ST magnitudes because the mean flux density is not the same as the actual flux density at the pivot wavelength, and that is implicitly being assumed in the equations to determine the AB and ST magnitudes.  The deviations will be particularly serious for the Guider because of the wide variation of flux density from 0.5 to 5.5 microns.  One should avoid using AB or ST magnitudes for the Guiders if one wants accurate count rates.

KevinVolkSTScI, 2018-03-27:
I made another revision of the guider throughput files.  I had neglected to include the fine focus mirror in the throughput values.  They should be correct now.

bhilbert4, 2018-04-02:
FGS imaging mode is now functional in my `add-fgs` branch. I've also added the new NIRISS throughput files. I have one more detail to look into before submitting a PR to bring into `master`. There are some pixels that have zero signal in my simulated test case. I want to see where these are coming from.

bhilbert4, 2018-07-13:
FGS simulation added via PR #107

closed 2018-07-13 by bhilbert4