spacetelescope/mirage #68: Phase out use of `observationlist.yaml` (?)

laurenmarietta, 2018-03-27:
I very well might be mistaken here, but in my use of the simulator, I have found that the generation of the `observationlist.yaml` file seems to be a non-essential step. I have been using `scripts/write_observationlist.py` to automatically generate these `.yaml` files from APT XML and pointing files. 

_If_ there are no cases where including the `observationlist.yaml` provides something that cannot otherwise be parsed from APT output (and please let me know if so!), it seems like it might be more efficient to skip this step, and go straight from the APT output to `.yaml` generation.

bhilbert4, 2018-03-28:
The motivation behind using the observation list yaml file was for cases where there are multiple filter pairs within a given observation, implying that the user would want to use multiple point source (or galaxy, or extended) catalogs within that observation. Using the observation list yaml file was the only way I could think of to be in this situation and still allow the simulator yaml files to be produced in an automated way with the correct catalog names. Have you been able to create input yaml files for this case without using the observation list?

laurenmarietta, 2018-03-29:
Ah, I see! No, I've only been running things in the F212N and F480M bands, since those are the bands that will be used for basically all of commissioning. So I've only ever had two catalog files (SW and LW) per observation.

Then I will leave the observation list as it is. If I happen to think of a way to easily incorporate different catalog files within the same observation (which I kind of doubt), I'll come back to it.

Thanks for pointing this out!

closed 2018-06-26 by laurenmarietta