spacetelescope/mirage #73: Implement APT interface for NIRISS

Johannes-Sahlmann, 2018-03-28:
Enhance the yaml_generator script (and dependents) to convert the content of an NIRISS APT file into simulator input files.

bhilbert4, 2018-07-19:
@Johannes-Sahlmann, I just tried running the `yaml_generator` on a NIRISS APT file and it crashed. It looks like the only NIRISS APT template that or works for is `NIRISSExternalCalibration`. Do I need to do something special in APT to access this template? Or is it out of date? All I see in APT now are templates for Imaging, WFSS, SOSS, and AMI.

Johannes-Sahlmann, 2018-07-19:
I think I only tested this for a `NIRISSExternalCalibration` APT program. You have to set the proposal category to ENG to have access to it.

bhilbert4, 2018-07-19:
Ok thanks. I was able to get to the `NIRISSExternalCalibration` template. It looks to be more complicated than I need for my test. At some point we should add at least the `Imaging` template, although when that happens depends I guess on how badly people need it. I'm just running a single simple test of full frame vs subarray, so maybe that's not enough of a driver to make this a high priority.