spacetelescope/mirage #75: Handling reference files

laurenmarietta, 2018-03-28:
Currently, the simulator is looking for NIRCam, NIRISS, and FGS reference files in both of the following directories (and maybe others?): 

- `/ifs/jwst/wit/nircam/nircam_simulator_data/`
- `/ifs/jwst/wit/niriss/nircam_ramp_simulation_files/`

Standardizing how reference files are organized, located, and named would serve to more easily keep the simulator references up-to-date, and would also make it easier to expand to instruments besides NIRCam, as is being discussed in issue #62.

I don't know much about reference files or CRDS, but another question: should we ultimately aim to be getting the most recent versions of reference files from crds, in `/grp/crds/cache/references/jwst`?

bhilbert4, 2018-03-28:
I agree. My hope was that once we have in hand all of the reference files for the three instruments, that we could place them all in a single directory tree somewhere. That will also make it easier to create tarballs that unpack in a nice way for external users.

As for CRDS files, I do think it would be nice to integrate with CRDS. I initially held off on that because early advice from a NIRCam IDT member was that they preferred the simulator to be independent of the JWST cal pipeline. So at that point the repo had to carry copies of reference files. I think that eventually having at least the option to query CRDS and download the appropriate reference files if they don't already exist locally would be good.

Eventually, I think that the way to go is to have users set a (e.g.) MIRAGE_REFS environment variable that points to the top level of the directories containing the reference files. Then this environment variable can be used in the input yaml files rather than full paths, if desired.

laurenmarietta, 2018-03-28:
Great! Setting an environment variable sounds like a perfect idea, once all the files have been created and collected.

How is it that using CRDS reference files creates a dependency on the calibration pipeline? From poking around a little on CRDS docs, it seems like downloading reference files locally isn't connected to any pipeline processes (but I would not be surprised if I had missed or misread something).

bhilbert4, 2018-03-28:
It's not connected to the pipeline, but my thought was that if the request was to be independent from the pipeline, then being independent from CRDS would also be true. 

Nor, who put together the disperser tool for the WFSS mode, wanted to see a function added that would download the necessary reference files at run time if they were not present locally. I think that's a good long term solution. Reducing the size of the collection of reference files would make everything easier, at the expense of having to wait for files to download during some runs of the simulator.

bhilbert4, 2019-04-30:
#295 contains code to download the reference file library to a user's machine and tell the user how to set the MIRAGE_DATA environment variable. This is the final step in making Mirage truly public.

Integrating with CRDS is still on the To Do list.

bhilbert4, 2019-08-14:
Returning to the topic of CRDS integration:
This should be relatively easy to implement now. It'd be nice to have this implemented prior to merging #295 and making everything truly public, so that we don't have to carry a copy of CRDS reference files in the mirage reference file library, which is hosted on Box. Then we don't need to worry about keeping those files up to date. And with CRDS integration, users download only the files they need, rather than a large library that they may only need parts of.

To use crds (which wouldn't technically be a new dependency since it is installed along with the jwst calibration pipeline):

- Set two CRDS environment variables:
- CRDS_SERVER_URL=https://jwst-crds.stsci.edu
- CRDS_PATH='/path/to/save/reffiles/into' (maybe integrate this with MIRAGE_DATA env var)
then:

```
import crds
parameters = {'INSTRUME': 'NIRCAM',
                         'DETECTOR': 'NRCB1',
                         etc, etc}
reffile_mapping = crds.getreferences(parameters, obervatory='jwst')
```
See: https://hst-crds.stsci.edu/static/users_guide/library_use.html#crds-getreferences

bhilbert4, 2019-08-14:
Reference files needed, and the parameters CRDS needs, to select the appropriate file. These must all be provided, in a dictionary, in the call to `getreferences` above.
```
bad pixel map:
NIRCAM: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS 
NIRISS: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS
FGS: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS, EXP_TYPE

distortion:
NIRCAM: INSTRUME, DETECTOR, EXP_TYPE, CHANNEL, FILTER, DATE-OBS, TIME-OBS
NIRISS: INSTRUME, EXP_TYPE, PUPIL, DATE-OBS, TIME-OBS
FGS: INSTRUME, DETECTOR, EXP_TYPE, DATE-OBS, TIME-OBS

gain:  
NIRCAM: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS
NIRISS:    INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS
FGS:        INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS

ipc:   
NIRCAM: INSTRUME, DETECTOR, DATE-OBS, TIME-OBS
NIRISS:   INSTRUME, DETECTOR, DATE-OBS, TIME-OBS
FGS:       INSTRUME, DETECTOR, DATE-OBS, TIME-OBS

linearity:  
NIRCAM: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS
NIRISS: INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS
FGS:      INSTRUME, DETECTOR, SUBARRAY, DATE-OBS, TIME-OBS

pam:  
NIRCAM: INSTRUME, DETECTOR, FILTER, PUPIL, EXP_TYPE, DATE-OBS, TIME-OBS
NIRISS:    INSTRUME, DETECTOR, FILTER, PUPIL, EXP_TYPE, DATE-OBS, TIME-OBS
FGS:         INSTRUME, DETECTOR, DATE-OBS, TIME-OBS

saturation: 
NIRCAM: INSTRUME, DETECTOR, DATE-OBS, TIME-OBS
NIRISS:    INSTRUME, DETECTOR, DATE-OBS, TIME-OBS
FGS:        INSTRUME, DETECTOR, DATE-OBS, TIME-OBS

superbias: 
NIRCAM: INSTRUME, DETECTOR, READPATT, SUBARRAY, DATE-OBS, TIME-OBS
NIRISS:   INSTRUME, DETECTOR, READPATT, SUBARRAY, DATE-OBS, TIME-OBS
FGS:       INSTRUME, DETECTOR, READPATT, SUBARRAY, DATE-OBS, TIME-OBS
```