spacetelescope/mirage #79: FITS import missing from unlinearize.py?

aliciacanipe, 2018-03-30:
I get an error that the FITS import is missing in unlinearize.py. 

/ifs/jwst/wit/witserv/data7/nrc/acanipe/simulated_images/newest_version/nircam_simulator/nircam_simulator/scripts/unlinearize.py in unlinearize(image, coeffs, sat, lin_satmap, maxiter, accuracy, robberto, save_accuracy_map, accuracy_file)
     74         devcheck = np.copy(dev)
     75         devcheck[i2] = -1.
---> 76         h0 = fits.PrimaryHDU()
     77         h1 = fits.ImageHDU(devcheck)
     78         hl = fits.HDUList([h0,h1])

NameError: name 'fits' is not defined

bhilbert4, 2018-03-30:
Interesting. This file hasn't been touched in months. It looks like it is missing the fits import, but fits is only used when there are pixels that fail to unlinearize correctly. I'll add in the fits import. But which linearity coefficients are you using?

bhilbert4, 2018-03-30:
Strange. I'm sure I've used the A1 coefficients before without a problem. 

Fits import statement added with PR #80. You should now get an output fits file with the accuracy map that comes out of unlinearize if there are pixels that don't unlinearize correctly.

aliciacanipe, 2018-03-30:
Okay, thanks! I'll see if I can sort out why my simulator case is having an issue when others aren't.

closed 2018-03-30 by bhilbert4