spacetelescope/mirage #83: Brainstorming Improvements to PSF Library/Generation

laurenmarietta, 2018-04-03:
After discussing the current way the simulator incorporates PSFs with Marshall, we feel that multiple collaborators should come together to think about the PSF library in the following contexts:
- The limits of having a discrete, 5x5 sub-pixel grid onto which stars are placed (especially in the context of precision astrometry)
- Redundancy and symmetry in the current organization of the library
- Benefits (and challenges) of referencing one single oversampled PSF and interpolating as appropriate, instead of querying pre-computed PSFs from files
- More complicated versions of interpolation (convolution?)
- Generally, balancing computation time with disk space

Johannes-Sahlmann, 2018-04-04:
I agree with this assessment and I would be happy to contribute to this effort.

Note that NIRISS is using a ~10x10 sub-pixel grid, but even basic astrometric source extraction is limited by that resolution very quickly.

laurenmarietta, 2018-07-17:
Notes from June 26th meeting with Bryan Hilbert, Marshall Perrin, Matteo Correnti, Armin Rest, Lauren Chambers, and Shannon Osborne regarding improving MIRAGE PSFs: 
[180626 Meeting Notes - PSFs in MIRAGE.pdf](https://github.com/spacetelescope/mirage/files/2202353/180626.Meeting.Notes.-.PSFs.in.MIRAGE.pdf)
 
This particular discussion was limited to NIRCam, but I imagine a lot of the ideas could be extended to NIRISS?

I am about to create a couple new issues based on that discussion.

laurenmarietta, 2018-08-07:
Closing now, as this has been addressed in subsequent issues.

closed 2018-08-07 by laurenmarietta