spacetelescope/mirage #84: Add support for input magnitudes in Vega system

Johannes-Sahlmann, 2018-04-05:
Add the option to specify the input source magnitude in the Vega systems (in addition to AB and ST mag).

@KevinVolkSTScI has some of the necessary conversion relationships, at least for NIRISS.

KevinVolkSTScI, 2018-04-05:
The A0V magnitude zero point count rate values for the NIRISS filters are in the file niriss_zeropoint_values.out, listed as the Mag column (see below), so these values should be available in the code when it reads the file.  One just needs to agree on what label to use in the input point source or extended source files to direct the code to use these values rather than the AB or ST ones.  I guess it could be labeled as Vega magnitudes, although that is slightly inaccurate.  Only absolute calibration pundits are likely to worry about the details involved.

# Filter Module    Mag       ABMag     STMag     PHOTFLAM      PHOTFNU    Pivot Wave.
F090W      N     27.20216  27.71256  28.79764  1.098931e-20  2.985403e-31  0.9025
F115W      N     27.00896  27.79268  29.40326  6.291057e-21  2.773018e-31  1.1495
F140M      N     25.74700  26.87095  28.91579  9.856256e-21  6.481081e-31  1.4040
F150W      N     26.44136  27.66345  29.84236  4.198384e-21  3.123538e-31  1.4935
F158M      N     25.61290  26.94171  29.24571  7.273480e-21  6.072129e-31  1.5820
F200W      N     26.03498  27.75178  30.55722  2.173399e-21  2.879494e-31  1.9930
F277W      N     25.44799  27.77173  31.28759  1.109150e-21  2.827052e-31  2.7643
F356W      N     24.99357  27.83384  31.91908  6.200033e-22  2.669862e-31  3.5930
F380M      N     23.13275  26.11887  30.34010  2.654521e-21  1.295627e-30  3.8252
F430M      N     22.66378  25.88038  30.34749  2.636527e-21  1.613895e-30  4.2838
F444W      N     24.45965  27.72568  32.26452  4.510422e-22  2.949529e-31  4.4277
F480M      N     22.54345  25.99384  30.71488  1.879638e-21  1.453751e-30  4.8152

KevinVolkSTScI, 2018-09-10:
This option is back in the code.  It certainly works for NIRISS and the Guiders.  I have not tried testing it for NIRCam.  I think this issue can be closed, but I will leave it for Johannes to judge this.

closed 2018-11-09 by Johannes-Sahlmann