spacetelescope/mirage #90: The "stmag" option in the source catalogue files does not seem to be working

KevinVolkAlt, 2018-05-14:
Testing of the "stmag" option for a point source list showed that this option did not work for me on the witserv3 machine.  I found that the condition used to test whether to switch from abmag to stmag was never returning True.  In fixing this I also added an option to use the "vegamag" string to enable use of the vegamag column in the NIRCam_zeropoints.list or niriss_zeropoints.list file.  Hence the addition of the normal magnitudes should work for both instruments.  I have only tested the change with NIRISS.

KevinVolkAlt, 2018-05-15:
I posted the code in branch magnitude-correction (I hope...it seemed to work from my computer).  The branch also includes the fix for the frame time.

KevinVolkSTScI, 2018-09-10:
I believe that this is fixed now and so I am closing this issue.

closed 2018-09-10 by KevinVolkSTScI