spacetelescope/mirage #91: When simulating a subarray the stars were not put in the right place

KevinVolkAlt, 2018-05-14:
The code does not generally locate stars correctly from a point source list when a subarray is being simulated, at least for NIRISS (but probably for NIRCam as well).  This was found to be due to having point source positions relative to the subarray in the code, whereas the RADectoXY routine produces return values of (x,y) for the pixel in the full frame coordinates.  One needs to subtract the self.subarray_bounds[0] and self.subarray_bounds[1] values to get pixel positions relative to the sub-array.  Related to this, the XYtoRADec routine needs input pixel values in the full frame but the code is passing values relative to the subarray so the RA/Dec output values are incorrect.  One in that case needs to add self.subarray_bounds[0] and self.subarray_bounds[1] to the pixel values that are being passed into XYtoRADec.  I have corrections for these problems ready to post when I get the account issues straightened out.

bhilbert4, 2018-05-14:
Hmm, interesting. I know I've checked full frame vs subarray simulations before using the same RA, Dec values and had sources show up where I expected. Maybe I was checking some special case or something. One of the things I want to work on when I get back is to add a proper testing module to the repo that uses Jenkins to automatically run tests each time the code is updated.

KevinVolkAlt, 2018-05-15:
The only case I can think of that would work is a subarray where self.subarray_bounds[0] and self.subarray_bounds[1] are equal to zero.  It happens that none of the NIRISS subarrays are of that sort in the DMS coordinates, although several of them are that way in the detector coordinates.

Johannes-Sahlmann, 2018-05-15:
@bhilbert4 do you have a strong preference for Jenkins compared to Travis? I am using travis for https://github.com/spacetelescope/pysiaf and I could try to set it up for mirage.
Let me know.

bhilbert4, 2018-05-16:
I don't have any preference between Travis/Jenkins. I think output file sizes may be a problem for testing on github, although I'm not positive.

bhilbert4, 2018-07-19:
I just did a quick check for NIRISS and NIRCam. For NIRCam I made a full frame and a SUB160 exposure, both with RA, Dec set to (4., 45.), and both using a point source catalog with a single source at (4, 45). So I expect the source to show up at the reference location in each of the exposures, which it did.

Repeating the same thing for NIRISS, using full frame as well as NIS_AMI1 and NIS_AMI4, the source did not show up in the subarray exposures. Very strange. I'm trying to think what code NIRCam is using that NIRISS is not, or vice versa that would result in different behavior.

KevinVolkSTScI, 2018-07-19:
My test run of the code is still running hours later, so I really am not sure what the problem is with the machine I am using.  I really have to wait until i can ssh to witserv3 to look at this again.  But the fix I posted before should still apply here so you might look at the self.subarray_bounds[0] and self.subaray_bounds[1] values for NIRCam as opposed to NIRISS to see if there is any difference.  The SIAF files may be treating things differently for the two cases, although it is difficult for me to imaging why it would be so.

KevinVolkSTScI, 2018-09-10:
This has been fixed (although it seems to have reappeared when trying to use pysiaf to get the aperture information...).  I am closing this issue.

closed 2018-09-10 by KevinVolkSTScI