spacetelescope/mirage #100: Update frame time calculation for NIRCam

bhilbert4, 2018-06-25:
According to JDOX page, it looks like the full frame exposure time matches that for NIRISS. Make sure the same is true for subarrays. If so, we can simply adopt the NIRISS changes.

bhilbert4, 2018-06-25:
Note that changes need to be made in catalog_seed_image.py as well as obs_generator.py

bhilbert4, 2018-07-25:
I finally have a moment to look into frametime calculations again. I want to update the NIRCam frame time calculation because it appears to be slightly off at the moment. Looking through some functions that John Stansberry gave to me, it looks like I was using the wrong value for the rowpad term when in full frame mode. (Using 2 when it should be 1). 

Comparing the expressions used, I see now that the expression is the same for all three instruments, with one small difference: since the fast and slow readout directions for NIRCam are 90 degrees different from those on NIRISS, the x and y terms are flipped between them. The FGS fast readout direction matches that of NIRCam, so technically FGS should use the same equation as NIRCam, right?

So in order to clean up this function, my thought is that I will re-write the equation for NIRCam/FGS to be the same as that for NIRISS, but with the terms for the x and y dimensions flipped. @KevinVolkSTScI , does that make sense?

NIRISS: 
frametime = (pad2 + (ydim/n_amps + colpad) * (xdim + rowpad)) * 0.00001

NIRCam/FGS: 
frametime = (pad2 + (xdim/n_amps + colpad) * (ydim + rowpad)) * 0.00001

bhilbert4, 2018-07-25:
I also plan to move this function into the `utils` folder since it is used in both `catalog_seed_image.py` and `obs_generator.py`

KevinVolkSTScI, 2018-07-25:
Yes this should work.  For FGS I think we are only going to simulate full-frame images so it does not matter whether x and y are transposed since both are always 2048 pixels.  I am not 100% certain which way is the fast read direction in FGS and whether it is the same for both detectors in the DMS orientation.  I will ask Pierre Chayer just to be sure.  Then if we ever allow sub-arrays to be used for FGS we will get the right frame time.

bhilbert4, 2018-07-25:
That sounds good. I was going by my printout of the JWST focal plane, which shows both FGS detectors with the fast readout in the vertical (V3) direction, just like NIRCam. 

Actually now that I check, it looks like FGS has posted its own frame time equation on JDox (and specifically talks about subarrays):
https://jwst-docs.stsci.edu/display/JTI/Fine+Guidance+Sensor%2C+FGS

KevinVolkSTScI, 2018-07-25:
I see that some CAL sub-arrays are mentioned on the JDox page.  We should add these to mirage.  I will get the information from Pierre or Shannon and add these to the guider subarrays file so that the FGS people can simulate these CAL observations as needed.

laurenmarietta, 2018-07-26:
Bryan is right that the fast readout is in the V3 direction. (Attaching a figure that shows the various FGS coordinate frames and readout directions.)
[image]

KevinVolkSTScI, 2018-07-26:
The above shows the fast read direction along science y = DMS y which is the same as NIRISS and the opposite of NIRCam.

laurenmarietta, 2018-07-26:
Also, @shanosborne provided me with the attached file listing the CAL subarrays and their parameters. The table doesn't include the V2 and V3 references, but that information is in the FGS SIAF.
[FGS_Cal_Subarray_parameters_v2.xlsx](https://github.com/spacetelescope/mirage/files/2232208/FGS_Cal_Subarray_parameters_v2.xlsx)

bhilbert4, 2018-07-26:
Thanks for the info. I'll have to update the printout on my wall so that FGS is in the proper orientation. I'll keep NIRISS and FGS grouped together in the frame time calculations.

bhilbert4, 2018-07-27:
Solved, via PR #128

closed 2018-07-27 by bhilbert4