spacetelescope/mirage #108: Check FilterConfig use in observation.yaml file

bhilbert4, 2018-06-28:
In my test, yaml_generator.py wanted there to be a separate FilterConfig for every dither of every exposure. It probably makes more sense to require a separate FilterConfig for each exposure. Filters are guaranteed to be the same across dither poisitions, and it seems like source catalogs should be as well.

Johannes-Sahlmann, 2018-10-17:
I am a bit uncertain about the meaning and use of `FilterConfig{:d}` and I am running into trouble with it in `AptInput.add_observation_info` (related to PR #172).
Did you conclude on the question raised in this issue?

bhilbert4, 2018-10-17:
FilterConfig specifies the input catalogs to be used in the SW and LW channel filters for each exposure in the APT file. Obviously this is only necessary for NIRCam. I should re-run this test, but when I made a simulated dataset last week, there was no problem having one FilterConfig for each exposure.