spacetelescope/mirage #117: Develop an interface to new NIRCam PSF Library

laurenmarietta, 2018-07-17:
As discussed in #83, connect MIRAGE to the new PSF library (#116).

This will, at the most fundamental level, require:

1. Providing the needed parameters (SCA, PSF location, filter) to query the library
2. Querying the library
3. Interpolating four nearest PSFs from the library to create a PSF with desired spatial variations for the given SCA and PSF location
4. Interpolating that resulting PSF to the desired sub-pixel location and detector sampling rate

bhilbert4, 2018-07-17:
As Marshall mentioned, it probably makes sense to first refactor the current code to use the ePSF framework with the current library.  I think this will generally help to make the code cleaner and simpler, and hopefully easier to drop in the new libraries when they are ready. Do you agree?

laurenmarietta, 2018-07-17:
Ah, right, yes I definitely agree with that as an intermediate step. 

So that would mean figuring out how to interpolate the different PSFs that are in the library now, then?

bhilbert4, 2018-07-17:
Possibly. I was thinking more of just the process of selecting the right file, loading it into the ePSF model, normalizing to the right flux level, and adding the flux to the seed image. The current PSFs are not oversampled, so interpolating probably wouldn't give good results. But maybe it does make sense to focus on just the PSFs that are (e.g.) centered in the pixel, and set up the framework to do the interpolation for the intra-pixel location, with the idea that the new library will function in that way?

laurenmarietta, 2018-07-19:
Oh, okay! That sounds good to me. You're right that the interpolation bit would be (I anticipate) the most difficult, and also wouldn't be accurate with the current library. Maybe worthwhile trying to figure it out anyway, at the end.

But what exactly do you mean by "loading it into the ePSF model"? Are you talking about the `EPSFModel` class? @shanosborne and I have been having some trouble wrapping our minds around the idea of the `EPSFModel` and how it differs from just a normal `numpy` array...

bhilbert4, 2018-07-19:
My take from reading over the `FittableImageModel` class documentation is that for this intermediate step, the `EPSFModel` wouldn't be much more than a `numpy` array. `Mirage` would find the right PSF file from the library, load the PSF into an instance of `EPSFModel`, and then would do something along the lines of:
`seed_image += psf.evaluate(x=x, y=y, flux=fluxlevel, x_0=xxi, y_0=yyi)`, where x and y are the 2D lists of pixel indexes covered by the PSF, fluxlevel is the total signal to scale it to, and x_0, y_0 is the pixel location of the center of the point source. While still using the old library though, it seems like x_0, y_0 would always be the center of the nearest pixel, since the psfs themselves are already interpolated for sub-pixel centering. 

I think the hard part of this is just going to be going through the relatively ugly current code, collecting together the relevant lines into functions that make sense, and then replacing the ugly code with new function calls.

I feel like I'm on the hook for that, since I put the original ugly code in place. Does that make sense? I think that's more or less what Marshall was talking about a couple weeks ago.

laurenmarietta, 2018-07-20:
Okay, so the `EPSFModel` class can be used to place the input PSF on the array where we want it, and to alter the flux... good to know. I should probably spend some time playing around with that class to better understand it.

If you want to make this update, feel free, but I also don't mind! I don't remember that part of the code being particularly messy.

bhilbert4, 2018-07-20:
I was thinking about this last night, in terms of the spatially varying PSF. Reading back over Larry's email, it does seem like the `EPSFModel` is limited at the moment in that it can do the sub-pixel interpolation to produce a PSF, but it can't handle the PSF that varies across the detector. There's no way to place our grid of PSFs from `WebbPSF` into a single instance of `EPSFModel` and have it produce a PSF for anywhere on the detector in a single step.

The best I can think of at the moment would be to build our own interpolator to produce an oversampled PSF at a given detector position, and then place that PSF in an instance of `EPSFModel`, and then have that do the sub-pixel interpolation and subsampling. At that rate, I'm not sure if we're trying too hard to shoehorn `EPSFModel` into the current code?

laurenmarietta, 2018-07-20:
Oh yes, we definitely will still have to work some magic outside of the `EPSFModel` to interpolate the four fiducial PSFs we query out of the library. (Though I think Larry said that he has a long-term goal to include that type of interpolation in the model, it's not there yet.)

I am thinking that if the `EPSFModel` can do the sub-pixel interpolation and flux conversion for us, that is still pretty helpful! At least easier than building a second interpolator to apply the sub-pixel position, and also writing in our own function for the flux. But if you think there would be benefits to not using it, we certainly don't have to.

bhilbert4, 2018-07-23:
PR #122 addresses the beginning of this work.

bhilbert4, 2018-08-03:
Now that we have a library generator, via #136 it seems like we can push forward with this. I think it'd be great to set things up so that the code is not tied to a specific library. A user can point `mirage` to their library that has been created at some oversampling factor with some number of fiducial PSFs, and `mirage` will be able to find the correct file, read it in, do the interpolation for detector position, populate the `fittableImageModel` or `EPSFModel`, and everything will be ready to go. Assuming we can get it working in that way, we would no longer have to provide a PSF library with the set of reference files for `mirage` (although maybe we still should, for convenience?).

laurenmarietta, 2018-08-06:
I definitely think that making any library query code generic is a great idea! And I'm hopeful that @shanosborne 's naming schema will make the query particularly easy.

I agree that it is a good idea to still have kind of a "standard" library that users can access located in the `mirage` repo on central stor. Just because we have yet to nail down how long it might take to make a library, and having an easily accessible one already generated that anyone can get to feels like a good thing to me! (For now I know Shannon created a test library that is available at `$MIRAGE_DATA/nircam/test_webbpsf_library`)

bhilbert4, 2018-08-07:
From @larrybradley, talking about developing an interpolation algorithm for position-dependent PSFs:

"This is something that we also want to do with ePSF models to handle geometric distortion.  My first thought is that simple bilinear interpolation of the 4 reference PSFs bounding the arbitrary (x, y) position might be sufficient.  It would be fast -- all you need is the coefficients calculated from the (x, y) position and then do the linear combinations on the reference PSF arrays (they would all need to have the same array shape)."

"Thinking about this more, one could do the bilinear interpolation to the nearest integer (x, y) position and then re-evaluate the PSF at a slightly shifted subpixel position.  The PSF models in photutils can do the re-evaluation for a subpixel shift (it involves another interpolation).  That might be good enough, but it would require lots of testing."

This is the method we had been talking about doing for `mirage`; interpolate once to get the position-dependent PSF centered on a pixel, and then leave it to ePSFModel to do the interpolation for sub-pixel position.

At the moment what I have is a call to scipy.interpolate.interp2d to create the interpolation function, which is made using all 16 of the reference PSFs across the detector. This needs to be done separately for each pixel. Then the interpolation function can be evaluated for any position on the detector. My initial thought was that for `mirage` we could create the interpolation functions early on in the code (because we only need one for a given detector/filter, and then we just need to evaluate those at the detector position for each source in the simulated image. But maybe it would be faster to create the interpolation functions separately for each source, using only the 4 nearest reference PSFs, and evaluate immediately? Would be worth testing, especially in the case where there are thousands of sources in the fov.

mperrin, 2018-08-07:
FYI, for high precision work on Hubble, Jay found that cubic rather than linear interpolation was necessary. But I think bilinear is a good place to start: both easier and faster, and we don't yet have hard evidence that higher order interpolation is necessary for our purposes. 

A particularly simple form of interpolation in this case would be a simple weighted sum of the 4 surrounding ePSF models, as @larrybradley describes. That will be faster than invoking interp2d a ton of times. 

I think we want something like an `EPSFModelArray` class which gets initialized once, and then does the appropriate bilinear interpolation by array operations for each source. The

bhilbert4, 2019-08-08:
This has all been solved via #134

closed 2019-08-08 by bhilbert4