spacetelescope/mirage #119: Create environment file to help others install mirage

bhilbert4, 2018-07-17:
This is related to a suggestion from Leonardo Ubeda, who has had trouble installing various dependencies (mostly healpy, used by jwst_backgrounds). He suggested an article showing how to install everything. But perhaps an exported yaml environment file would be easier, as we have done with jwql. Then we can include this in the repo.

bhilbert4, 2018-07-17:
I can do this, although I have a lot of extra junk in the environment that I'm using for mirage. I'd probably start over with a new environment that's contains only the necessary packages. Does anyone else have a cleaner environment that we could package up? @laurenmarietta @Johannes-Sahlmann 

Here are the commands to make the file, as posted in the wiki article for jwql:
1. activate the environment
2. conda env export > environment.yml
3. environment.yml contains a list of packages and versions of everything in the environment
4. To create this environment on a difference machine: conda env create -f environment.yml

Johannes-Sahlmann, 2018-07-17:
I also don't have a 'clean' environment to start with. If we choose the environment.yml solution, we could start with the default jwst pipeline environment?
Alternatively, I believe that many problems can be avoided by using the `install_requires` option in `setup.py`. I have to implement this anyway for Travis so I can give it a try and report back.

bhilbert4, 2018-07-17:
I'd be interested to see that.

bhilbert4, 2018-07-27:
For various reasons I had to uninstall and reinstall Anaconda today. That gave me an opportunity to start over in creating a relatively clean environment for `mirage`. I did that and created an environment yaml file, similar to what has been done for `jwql`. I wasn't able to use the environment yaml file to create a new environment though. It crashed when trying to install `grismconf` for some reason. It was looking for a version beyond 1.0, which is strange since 1.0 is the latest version. I'll have to poke around and try to figure out what's going on.

mperrin, 2018-08-14:
I'd like to install mirage and start learning to use it a little. If I wanted to test out your environment file for installing this, where can I find that? Let me know what the currently recommended way to proceed is.

bhilbert4, 2018-08-14:
I tried making an environment file for `mirage`, but was having trouble getting it to work. It kept getting stuck on one of Nor's grism packages, looking for a version that doesn't exist. I still haven't figured out what's going on there. In the meantime installation is manual, and dependencies are listed in setup.py. I'll also send you an email with the local setting for the environment variable `MIRAGE_DATA`.

laurenmarietta, 2018-08-30:
Just wanted to bump this - working with a loaner laptop has made me realize how helpful having an `environment.yaml` would be!

bhilbert4, 2018-08-31:
Functional environment file produced and tested, and introduced in #158.

closed 2018-10-02 by bhilbert4