spacetelescope/mirage #124: PSF positions are not exactly as requested due to code moving the peak of the PSF to the center of the PSF image

KevinVolkSTScI, 2018-07-24:
The WebbPSF input images for NIRISS do not always have the peak pixel of the 301x301 pixel PSF image at pixel (151,151) [using 1-based pixel numbering).  This characteristic of the PSF files is as it should be in cases where the star position is offset from the center of the pixel in the middle of the PSF image.  In the code within the catalog_seed_image.py script when a PSF image is read in the peak pixel is moved the center of the PSF stamp image (i.e. lines 1798 to 1800 where nxshift and nyshift are found).  When this is done, star positions may get moved off from where they should be by 1 pixel in x or in y (or both).  One should not offset the peak of the PSF file from WebbPSF in this way, in my opinion.  I had to take this shift out of the code to get a single star at the reference (RA,Dec) to be put at the correct positions in the various NIRISS sub-arrays.

bhilbert4, 2018-07-24:
I see what you're saying. I think that makes sense. I had originally put those lines in there in case someone tried to use webbpsf outputs where the source was shifted away from the center by multiple pixels. But looking back on it now, this doesn't seem like a reasonable case to worry about.

As part of the PSF refactoring that I've been working on for the last couple days, I updated this section of the code, and in fact replaced the lines you mentioned with lines where `nxshift` and `nyshift` are simply set to the center of the array containing the PSF. One thing I'm not sure about though is if updated need to be made in the case of PSF arrays with odd numbers of rows/columns compared to even. Could you check the updated code (in PR #122)  using your NIRISS subarray cases? I'm curious to hear if there is still a shift since the PSF files are 301x301, which means `nxshift` and `nyshift` will be 150 (in 0-indexed coordinates).

KevinVolkSTScI, 2018-07-24:
I have been running through the NIRISS subarrays systematically with the following lines in place in the code to make the nxshift and nyshift values 150 by default for the NIRISS PSF files

            nxshift = int((psfdims[1]-1)/2)
            if a_in == -0.5:
                nxshift=nxshift+1 
            nyshift = int((psfdims[0]-1)/2)

and so far everything is in the right position.  However, note the x offset by 1 pixel in the case where a_in is -0.5; it is needed for the NIRISS PSF library, but I am not able to tell you why it is needed.  There must be something in WebbPSF that I am not doing properly which affects just the x values at the pixel edge.  In any case, this test is consistent with what you are doing aside from the extra offset of nxshift.

KevinVolkSTScI, 2018-07-24:
I will try getting the revised code and running the same set of tests of the NIRISS sub-arrays through that branch in the next day or so, if nothing else comes up.  I had to make some changes to niriss_subarrays.list for the testing I am doing right now because there is a discrepancy between the SIAF pixel values which are such that the center of pixel x=1 y=1 with 1 based numbering is (1.0,1.0) whereas in mirage the pixel center is (0.5,0.5) and position (0,0) is the corner of the pixel.  Then I found the offset of 1 in x that is mentioned in the previous comment.  I am still working through the different NIRISS sub-arrays at the moment with the mean branch code.

KevinVolkSTScI, 2018-07-31:
By the way, the same x shift by 1 pixel when a_in is -0.5 applies to the Guider.  The same fix works there as well.  In the Guider case all the reference points for the arrays are in the corner of a pixel so it affects all the positions.

bhilbert4, 2018-08-17:
While testing some of the new pysiaf functionality in my branch, I've had a chance to look closely at PSF locations. I've noticed a couple of issues. 

First, the construction of the filename from the PSF library seems to be incorrect. It looks like I built that using the assumption that x,y = (0.5, 0.5) would be centered in the pixel, which is inconsistent with the coordinate system definition elsewhere. In looking at the PSFs in the actual library files, I see that integer values are centered on a pixel, while (0.5, 0.5) corresponds to the edge of a pixel, as expected.  What this essentially means is that the file read in for a PSF at a given x,y location contains a PSF that is shifted by 0.5 pixels from what it should be. As far as I can see, this happens for every source in all cases.

The second issue I noticed is that the PSFs in the NIRISS library shift in the opposite direction from what I expect along the x axis only. For example, comparing the file for (0, 0) and (+0.1, 0), the PSF is shifted to the left in the (+0.1, 0) file. I expect it to shift to the right for increasing x value. Along the y axis, the PSFs shift was I'd expect (i.e. upwards for positive values).

So the question is, should I fix this or, given that the EPSF update is on the way and should solve all of these issues, should we just live with the code as is until then?

KevinVolkSTScI, 2018-08-23:
In the original rampsim.py code the pixel positions were calculated assuming that the pixel center is at (0.5,0.5).  Since the PSF files have names matching the IRAF/DS9 convention where the pixel center is at (0,0.) this is allowed for when going from the pixel positions the code calculates to make the PSF library file name. 

I can tell you that the PSF libraries that I have made for testing the code and the current code place the stars where they are supposed to be except for the case of a pixel position right on the left edge where there is the unexplained shift of 1 pixel compared to where the star should be.  I explicitly tested this for each NIRISS and Guider sub-array.  However Bryan may be looking at an out-of-date PSF library.  There was an issue that I had the star position flipped in x due to having the wrong sign in an arc-tan calculation in the code.  I fixed this and recalculated the PSF library for NIRISS but Bryan is probably using the first PSF library where the bug was present in the code that made the files given what he says in the comment.

bhilbert4, 2019-03-26:
Yikes. I totally forgot about this because it was wrapped up in the FittableImageModel stuff. As I was confirming source placement yesterday I noticed offsets between the FittableImageModel work (which now uses `photutils` `GriddedPSFModel`, and Mirage's master branch. I tracked it down to incorrect positions in the master branch because the wrong PSF library files were being chosen for a given subpixel position. 

I created a new branch and made a fix. In my tests, the master branch is now selecting the correct file from the PSF library and the GriddedPSFModel branch are both placing sources in the same locations, which agree with the positions in the input source catalogs. 

(The GriddedPSFModel seems to place the sources more accurately, with up to an order of magnitude lower offset than what the master branch does.)

But anyway, this quick fix for the master branch can be merged in soon. My tests all seem to give good results. The one question I have @KevinVolkSTScI is about the offset you put in for the case where the source is at a phase of -0.5. Is that still necessary? I just flipped through some NIRISS psf files and it looks like the -0.5 case has the psf where it should be (i.e. 0.1 pixels away from the -0.4 case).

```
   if a_in == -0.5:
        nxshift=nxshift+1 
```

KevinVolkSTScI, 2019-03-26:
I have to make an explicit test of this to see if it is still needed.  Before the -0.5 case was put 1 pixel off and this was a fix to override that.  One can hope that this is no longer needed.

bhilbert4, 2019-04-11:
@KevinVolkSTScI have you had time to think about this? I just tried running a NIRISS test and was not able to place a source such that the -0.5 PSF library file was ever chosen. Instead +0.5 was always chosen, which put the source in the correct location.

KevinVolkSTScI, 2019-04-11:
I have not taken the time to test this.  But if the sources are in the right position for the -0.5 case I suppose we should leave the code as is.  The biggest problem with the previous "patch" I did is that only the x values seemed to need correction and I never was able to determine why in that the code seems to treat x and y in exactly the same way.
I can try to test this today or tomorrow but which version/branch of Mirage should I use?

bhilbert4, 2019-04-11:
PR #280 contains the new code. The branch name is very generic: 'investigation'. 

I'd be interested to hear if you can get a source that uses the -0.5 case. I couldn't. Have you looked at the PSF stamps themselves to see if the problem is in there? I seem to recall a bug in webbpsf that was pushing NIRCam PSFs an extra pixel to the side in some cases.  

Also, I've been pushing hard on the branch that switches Mirage to use the griddedPSFModel (ePSF) rather than these libraries. The code is more or less ready. I just need to create the actual PSF library files via weebbpsf. Those changes will basically overwrite all of these changes, but I think it's probably still good to make sure this is working correctly, since it's a small fix.

KevinVolkSTScI, 2019-04-11:
As far as I remember the WebbPSF files had the source where it was expected to be for all the images, including the -0.5 case.  But it was a long time ago that I was looking at that.  

I would prefer to have a working version with the old PSF libraries around even when you switch to the gridded PSFs just because I know that it does what I want.

bhilbert4, 2019-04-11:
For what it's worth, in my tests with the griddedPSFModels, sources were being placed more accurately (relative to the coordinates specified in the source catalogs) by almost an order of magnitude compared to the current code. (Based on centroiding results from photutils) This was for a NIRCam griddedPSFModel library oversampled by a factor of 5, compared to the current case, where the NIRCam PSF library has a resolution of 0.25 pixels. But anyway, this is something we can get into in the PR for the new code. Or actually maybe I will create a new issue where we can discuss the details of the libraries.

KevinVolkSTScI, 2019-04-11:
I suppose it will all be good once I get some experience with the new libraries...I will need some instructions about how to make these for NIRISS and the Guider.

KevinVolkSTScI, 2019-04-11:
@bhilbert4 I ran a set of tests with a star being offset across the edge of a pixel both in pixel position input and sky position input and in all cases there were no problems with the positions in the output seed images.  So whatever is in the code now is working fine as far as I can see and can be posted.

bhilbert4, 2019-04-11:
Excellent! Thanks for testing.

PR #280 (2019-04-11): Choose correct pixel phase file from PSF library
This PR fixes a bug in `catalog_seed_image.py` where the wrong file was being chosen from the PSF library for a given pixel phase. The code was originally written such that a pixel phase of 0.5, 0.5 was defined as the center of a pixel. This is inconsistent with the definition used in `pysiaf` and the pipeline, where 0.0, 0.0 is the center of the pixel, and -0.5 and 0.5 are the edges.

This fix changes the pixel phase definition to agree with that used by Mirage's dependencies.

Closes #124 
@KevinVolkSTScI
closed 2019-04-11 by bhilbert4