spacetelescope/mirage #129: Re-organization of scripts

laurenmarietta, 2018-07-27:
It might be wise to discuss how all the `mirage` scripts are currently organized within the repo, and whether there is enough reason to re-organize them. At the moment, every script is within the `scripts` directory, but designing more structure into the repo could make it easier to understand the scripts' functions.

Here's one particular option I came up with for an alternate directory structure:
```
mirage/
    examples/
    tests/
    mirage/
        catalogs/
        config/
        apt/
            __init__.py
            apt_inputs.py
            read_apt_xml.py
        seed_image/
            __init__.py
            catalog_seed_image.py
            get_psf.py
            moving_targets.py
            segmentation_map.py
        dark/
            __init__.py
            dark_prep.py
        ramp_generator/
            __init__.py
            obs_generator.py
            unlinearize.py
        setup/
            __init__.py
            get_catalog.py
            write_observationlist.py
            yaml_generator.py
        utils/
            __init__.py
            polynomial.py
            read_fits.py
            read_siaf_table.py
            rotations.py
            set_telescope_pointing_separated.py
            utils.py
            yaml_update.py
        __init__.py
        imaging_simulator.py
        wfss_simulator.py
    README.md
    setup.py
```

In this case, something like `apt_inputs` would then be imported with a new statement: `from mirage.apt import apt_inputs`.

I am not sure where `blot_image.py`, `coord_transform_code.py`, `crop_and_blot.py`, `crop_mosaic.py`, `fits_seed_image.py` would be placed, since I have never used these. 

I am curious what others' opinions are!

bhilbert4, 2018-07-27:
I think in general it's a good idea. Your layout would make things more clear to people writing and using the code. 

That being said, I'm a little gun shy at the moment due to all the trouble I had when trying to rename things from `nircam_simulator` to `mirage`. It may be unrelated, but I've been having problems with `git` on my machine ever since then.

laurenmarietta, 2018-07-27:
That is certainly fair - I realized while thinking it through that this would mean another major overhaul to really the entire repo and every single import statement. Definitely a daunting task, and not a particularly urgent one. 

I just thought I might document the idea here so we could discuss, and maybe decide to make an update sometime in the future.

Johannes-Sahlmann, 2018-07-27:
Looks good to me. Only suggestion is not to create a directory names `setup` to avoid confusion with `setup.py`. Could be named `initialize/`.

I have made good experiences with pycharm's `refactor->rename/move` commands, which seems to take care of all the git stuff under the hood.

laurenmarietta, 2018-07-27:
I hadn't thought about the conflict with `setup.py` - thanks for that.

I would be happy to spearhead this process sometime, Bryan, if there is a lull in development where it wouldn't cause any massive merge conflicts with what others are working on. I haven't used pycharm before (though it sounds like I should!), but I have successfully reorganized a couple repos in the past.

bhilbert4, 2018-07-27:
You're welcome to it. :) The only development I have at the moment is the ePSFModel work. I just made a separate copy of those files outside of my repo because of my git troubles. I'm thinking about deleting my local copy of the repo and recloning just in case my .git files are screwy for some reason. Until I clear up these issues though, I don't think I'm going to be doing any development work on `mirage`. So if you're inspired, feel free to go for it.

laurenmarietta, 2018-07-30:
Okay, I will try to get this done this morning.

For the files that I wasn't sure where to place (blot_image.py, coord_transform_code.py, crop_and_blot.py, crop_mosaic.py, fits_seed_image.py), do you have an idea for where they might go? Do they still need to be in the repo if they aren't being used? (Or is it just me who isn't using them....)

bhilbert4, 2018-07-30:
Those files are used when creating a seed image from a large distortionless fits file (like an HST mosaic from the HLA, or something along those lines. So those can go into the `seed_image` directory.

laurenmarietta, 2018-07-31:
Closed via #130

closed 2018-07-31 by laurenmarietta