spacetelescope/mirage #131: wfss_simulator throws error

laurenmarietta, 2018-07-31:
When I was running `WFSS_simulator_use_examples.ipynb`, I ran into the following error while trying to run the `wfss_simulator` module:

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-19-e377bf38ce73> in <module>()
     19 #m.override_dark = dark_prep_output
     20 
---> 21 m.create()

~/TEL/mirage/mirage/wfss_simulator.py in create(self)
    102         disp_seed = Grism_seed(imseeds, self.crossing_filter,
    103                                dmode, config_path=loc,
--> 104                                extrapolate_SED=self.extrapolate_SED)
    105         disp_seed.observation()
    106         disp_seed.finalize(Back = background_file)

TypeError: __init__() got an unexpected keyword argument 'extrapolate_SED'
```

It seems like this is an error from within the `wfss_simulator` code, rather than a problem in the notebook?

bhilbert4, 2018-07-31:
My guess is that your version of NIRCAM_Gsim is old. Nor added the `extrapolate_SED` option several months ago.

laurenmarietta, 2018-07-31:
That fixed it. Thank you!

closed 2018-07-31 by laurenmarietta