spacetelescope/mirage #132: catalog_seed_image throws errors for moving targets

laurenmarietta, 2018-07-31:
When I was running `MovingTarget_simulator_use_examples.ipynb`, I ran into the following error while trying to run the `create_seed_image` module:

```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in readPointSourceFile(self, filename)
   1950         try:
-> 1951             gtab = ascii.read(filename)
   1952             # Look at the header lines to see if inputs

~/miniconda3/envs/OTECommSims/lib/python3.5/site-packages/astropy/io/ascii/ui.py in read(table, guess, **kwargs)
    350         # more useful traceback.
--> 351         dat = _guess(table, new_kwargs, format, fast_reader)
    352         if dat is None:

~/miniconda3/envs/OTECommSims/lib/python3.5/site-packages/astropy/io/ascii/ui.py in _guess(table, read_kwargs, format, fast_reader)
    496             reader.guessing = True
--> 497             dat = reader.read(table)
    498             _read_trace.append({'kwargs': guess_kwargs,

~/miniconda3/envs/OTECommSims/lib/python3.5/site-packages/astropy/io/ascii/core.py in read(self, table)
   1145         # Get a list of the lines (rows) in the table
-> 1146         self.lines = self.inputter.get_lines(table)
   1147 

~/miniconda3/envs/OTECommSims/lib/python3.5/site-packages/astropy/io/ascii/core.py in get_lines(self, table)
    292                 with get_readable_fileobj(table,
--> 293                                           encoding=self.encoding) as fileobj:
    294                     table = fileobj.read()

~/miniconda3/envs/OTECommSims/lib/python3.5/contextlib.py in __enter__(self)
     58         try:
---> 59             return next(self.gen)
     60         except StopIteration:

~/miniconda3/envs/OTECommSims/lib/python3.5/site-packages/astropy/utils/data.py in get_readable_fileobj(name_or_obj, encoding, cache, show_progress, remote_timeout)
    192                 timeout=remote_timeout)
--> 193         fileobj = io.FileIO(name_or_obj, 'r')
    194         if is_url and not cache:

FileNotFoundError: [Errno 2] No such file or directory: 'temp_non_sidereal_point_sources.list'

During handling of the above exception, another exception occurred:

OSError                                   Traceback (most recent call last)
~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in getPointSourceList(self, filename)
   1496         try:
-> 1497             lines, pixelflag, magsys = self.readPointSourceFile(filename)
   1498             if pixelflag:

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in readPointSourceFile(self, filename)
   1968         except:
-> 1969             raise IOError("WARNING: Unable to open the source list file {}".format(filename))
   1970 

OSError: WARNING: Unable to open the source list file temp_non_sidereal_point_sources.list

During handling of the above exception, another exception occurred:

NameError                                 Traceback (most recent call last)
<ipython-input-10-fcd6eb6fe9a1> in <module>()
      1 cat = catalog_seed_image.Catalog_seed()
      2 cat.paramfile = yamlfile
----> 3 cat.make_seed()

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in make_seed(self)
    143         if self.params['Telescope']['tracking'].lower() == 'non-sidereal':
    144             print('Creating signal ramp of synthetic inputs')
--> 145             self.seedimage, self.seed_segmap = self.non_sidereal_seed()
    146             outapp = '_nonsidereal_target'
    147 

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in non_sidereal_seed(self)
    548         # These will be stationary in the fov
    549         nonsidereal_countrate, nonsidereal_segmap, self.ra_vel, self.dec_vel, vel_flag \
--> 550             = self.nonsidereal_CRImage(self.params['simSignals']['movingTargetToTrack'])
    551 
    552         # Expand into a RAPID exposure and convert from signal rate to signals

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in nonsidereal_CRImage(self, file)
   1202             ptsrc.write(os.path.join(self.params['Output']['directory'], 'temp_non_sidereal_point_sources.list'), format='ascii', overwrite=True)
   1203 
-> 1204             ptsrc = self.getPointSourceList('temp_non_sidereal_point_sources.list')
   1205             ptsrcCRImage, ptsrcCRSegmap = self.makePointSourceImage(ptsrc)
   1206             totalCRList.append(ptsrcCRImage)

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in getPointSourceList(self, filename)
   1501                 print("Point list input positions assumed to be in units of RA and Dec.")
   1502         except:
-> 1503             raise NameError("WARNING: Unable to open the point source list file {}".format(filename))
   1504 
   1505         # File to save adjusted point source locations

NameError: WARNING: Unable to open the point source list file temp_non_sidereal_point_sources.list
```

It can't find the `temp_non_sidereal_point_sources.list`, even though that file exists and was created in the directory specified in `moving_target_test.yaml` (Observation/directory). This particular error goes away if I move the `temp_non_sidereal_point_sources.list` file that it created into the same directory as the notebook itself. 
I imagine I could fix this by telling `create_seed_image` to look in the specified output directory - is that reasonable?

But after that error, the following error is thrown:
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-9-fcd6eb6fe9a1> in <module>()
      1 cat = catalog_seed_image.Catalog_seed()
      2 cat.paramfile = yamlfile
----> 3 cat.make_seed()

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in make_seed(self)
    143         if self.params['Telescope']['tracking'].lower() == 'non-sidereal':
    144             print('Creating signal ramp of synthetic inputs')
--> 145             self.seedimage, self.seed_segmap = self.non_sidereal_seed()
    146             outapp = '_nonsidereal_target'
    147 

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in non_sidereal_seed(self)
    599                                                                         tracking_ra_vel=self.ra_vel,
    600                                                                         tracking_dec_vel=self.dec_vel,
--> 601                                                                         trackingPixVelFlag=vel_flag)
    602             # Multiply by pixel area map
    603             mtt_galaxies *= self.pam

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in movingTargetInputs(self, file, input_type, MT_tracking, tracking_ra_vel, tracking_dec_vel, trackingPixVelFlag)
    833                " the detector."))
    834         print("{} sources in original input catalog {}.".format(len(mtlist), file))
--> 835         indexes, mtlist  = self.remove_outside_fov_sources(indexes, mtlist, pixelFlag, 4096)
    836         print("{} sources in filtered input catalog.".format(len(mtlist)))
    837 

~/TEL/mirage/mirage/seed_image/catalog_seed_image.py in remove_outside_fov_sources(self, index, source, pixflag, delta_pixels)
   1766         x = source['x_or_RA']
   1767         y = source['y_or_Dec']
-> 1768         good = ((x > minx) & (x < maxx) & (y > miny) & (y < maxy))
   1769         filtered_sources = source[good]
   1770         filtered_indexes = index[good]

TypeError: unorderable types: Column() > float()
```
This error seems more complicated. Thoughts?

bhilbert4, 2018-07-31:
This seems like maybe it's a problem with the format of the catalog file? Maybe the file is being found, but since the format is wrong it isn't being read in? 

I'm confused about the Column vs float error though. Does the format of the catalog look ok?

laurenmarietta, 2018-07-31:
Hmm, I'm not certain about the catalog file. I kind of doubt it's a formatting issue, because the error goes away if I just move the file to a different location. And anyway, the `temp_non_sidereal_point_sources.list` file that is being generated by `create_seed_image` itself as a copy of the information in `moving_target_to_track.cat`. So it seems odd that it would be automatically generating a file with incorrect formatting?

I'll poke around in `create_seed_image` and see if I can get that first error to go away.

As for the Column vs. float error, it seems to be coming from the galaxies catalog. It looks okay to me... here is all that is in it:

```
#
#
# 
#abmag
#Columns 1 and 2 can be either x,y positions on the detector aperture (e.g.
#0,0 is lower left corner of the full frame of the subarray used for the 
#output) or RA,Dec location of the center of the source. If they are x,y
#positions, make the top line of the file '# position_pixels'
#
#radius is the half-light radius in pixels or arcseconds. If in pixels
#make the second line of the file '# radius_pixels
#
#pos_angle is the position angle of the semimajor axis, in degrees.
#0 causes the semi-major axis to be horizontal.
#
#x_or_RA_velocity is the velocity in the increasing x or RA direction
#in units of arcsec/hour. To specify units in the x-direction, place
#'velocity_pixels' (alone) in one of the top 4 lines of the file.
#
#y_or_Dec_velocity is the velocity in the increasing y or Dec direction
#in units of arcsec/hour. To specify units in the y-direction, place
#'velocity_pixels' (alone) in one of the top 4 lines of the file.
x_or_RA    y_or_Dec  radius  ellipticity  pos_angle  sersic_index  magnitude  x_or_RA_velocity y_or_Dec_velocity
23:59:59.9986 +00:00:00.0047  1.0      0.25        20           2.0  16.000        -0.5               -0.02
00:00:00.0003 +00:00:00.0216  1.5      0.5         79           1.2  16.000        -0.5               -0.02
23:59:58.6185 +00:00:32.1260  1.5      0.25        0            2.0  18.000        -0.5               -0.02
00:00:02.1327 +00:00:20.7322  1.5      0.25        70           2.0  17.000        -0.5               -0.02
```

laurenmarietta, 2018-07-31:
Okay, I was able to fix the first error with a pretty simple fix - it just needed to pass the absolute path of the temp file instead of just the filename to `getPointSourceList()`.

As for the second error, it seems I was just providing the wrong file in `moving_target_test.yaml` - my bad. I had provided `moving_galaxies.list` as the `galaxyListFile`, but changing that to be `moving_target_extended.list` made everything run smoothly.

Do you happen to know what `moving_galaxies.list` is used for, then? Should it stay in the repo?

bhilbert4, 2018-07-31:
Oh good. I'm glad it was an easy fix. `moving_galaxies.list`, which probably should actually be called `moving_sersic.list` or something like that, is used in the case where you are want to have resolved elliptical objects trailing through the background. Basically my thinking in adding this mode was for people who were simulating (e.g.) a binary KBO and wanted to have the "satellite" trailing through the background while focusing on the primary body. It's a pretty limited use case, but my background is outer solar system. :) There may be other use cases I haven't thought of as well.

laurenmarietta, 2018-08-01:
Gotcha! I will rename it to be `moving_sersic.list` and then put in a PR for the bug fixes.

So, out of curiosity, would that file be the `movingTargetSersic` parameter in the yaml?

bhilbert4, 2018-08-01:
Yep, that's right.

laurenmarietta, 2018-08-02:
Closed via #135

closed 2018-08-02 by laurenmarietta