spacetelescope/mirage #138: Guider2 PSFs should be flipped in y compared to Guider1 PSFs

bhilbert4, 2018-08-07:
The PSFs in Guider2 are simply flipped in y compared to those for Guider1. In order to decrease the number of PSF libraries needed, we could simply have a single PSF library for FGS, and flip the PSF in y when simulating Guider2.

laurenmarietta, 2018-08-07:
Can you include a screenshot to demonstrate what you mean? I am wondering whether the flip you are seeing is actually a result of the difference in coordinate frames between the two guiders.

I'm curious what @shanosborne will say when she gets back from vacation, but I don't think we want to just have one PSF for both guiders, as then we would lose the ability to take into account geometric distortion and field dependence, which are different in each detector.

bhilbert4, 2018-08-07:
I'll leave this to @KevinVolkSTScI to address, as he is the one who requested it.

KevinVolkSTScI, 2018-08-07:
Currently WebbPSF only calculates one PSF for the Guider.  The only difference between sending light to Guider 1 rather than Guider 2 is that a flat fold mirror is inserted into the beam.  This will not affect the distortion and such, all of which will happen prior to the fold mirror.  (One assumes that the fold mirror is flat and does not introduce distortions.)  I asked Pierre Chayer whether the flip mirror produces a flip in the PSF on the detector and he said yes it does.  It might be that he was incorrect in the regard, but for the moment I would like to put it in.

WebbPSF does not calculate the internal effects of the fold mirror and only calculates a single PSF as delivered to the Guider POM.

mperrin, 2018-08-07:
`Currently WebbPSF only calculates one PSF for the Guider.`
Sorry, that's not correct. We have separate models for FGS1 and FGS2, including their separate measured WFE data from ISIM testing.  The `.detector` attribute allows selection between FGS1 and FGS2. 

Here we need to be careful about coordinate systems. The fold mirror does introduce a difference in the raw detector coordinate systems ("DET" frame, in SIAF-speak). But DMS takes that into account when generating calibrated files, and flips as necessary. All DMS-output files have the same handedness in the output science data ("SCI" frame in SIAF). It's my understanding that is the case for FGS full-frame images, and I think for the subarray postage stamps too?

Johannes-Sahlmann, 2018-08-08:
To be explicit, the SIAF Science coordinate system and the DMS coordinate system are the same. 
The SIAF Detector coordinate system and the DMS `raw` coordinate system are generally **not** the same. pysiaf now includes code to transform between `sci`, `det` and `raw` coordinate systems: https://github.com/spacetelescope/pysiaf/blob/bf40d17a6056c2a50d0671625ff19bcdbaff08b3/pysiaf/aperture.py#L987

laurenmarietta, 2018-08-08:
Because I invariably find it helpful in conversations like this, here is a diagram of all the FGS coordinate systems for reference:
[image]

Johannes-Sahlmann, 2018-08-08:
The conversion from raw to sci happens before the data enter what we call the pipeline processing. See e.g. JWST-STScI-002566, SM-12 (page 8-2 and Section 8.7).

KevinVolkSTScI, 2018-08-08:
All I know is that the input files to the pipeline are in the DMS orientation (level 1b I think it is, before any processing steps in the pipeline), and that is what we are trying to produce with mirage in most cases.

laurenmarietta, 2018-08-08:
Yes, the FGS images that MAGIC creates are always in the raw frame, for the appropriate FGS.

I am curious, though - I generated PSFs for GUIDER1 and GUIDER2 in WebbPSF and there does seem to be a flip between the two of them. WebbPSF generates PSFs in the science/DMS coordinate frame, correct? So would we expect the flip to be present in WebbPSF products? Screenshot:
[image]

laurenmarietta, 2018-08-16:
Any updates here? @KevinVolkSTScI , have you been able to double-check with Pierre to determine whether the FGS PSFs do need to be changed in the library?

KevinVolkSTScI, 2018-08-23:
I have to look into this again now that I am back from vacation.  The fold mirror definitely flips the field but the DMS orientation takes it out again.  As Marshall noted that would mean that no flip of the WebbPSF files is needed, but I have to as Pierre again to try to be sure about this.

KevinVolkSTScI, 2018-08-27:
I have asked Pierre again and based on the discussion with him it appears that the PSF for the Guider should NOT need a flip between the Guider 1 and Guider 2 cases, in agreement with what  Marshall stated above.  Hence I am closing this issue.

closed 2018-08-27 by KevinVolkSTScI