spacetelescope/mirage #144: x, y conventions reversed in psf_library.py

bhilbert4, 2018-08-14:
The documentation in psf_library.py states that the data are stored in a 5D array with dimensions [detector, i, j, x, y], where i and j are position indexes across the detector, and x,y are the two dimensions of the PSF. However, tracing back through the code to the way it calls `webbpsff`, it appears that the dimensions should actually be listed as [detector, j, i, y, x]. That is, the conventional y dimension comes before the x dimension in both cases. If this is true (and you should double check my interpretation), it would be helpful if at least the documentation, if not the code, is changed to reflect this.

bhilbert4, 2018-08-14:
The DET_XY and DET_IJ header keywords should probably change as well.

shanosborne, 2018-08-17:
Yes you're right it should be [detector, j, i, y, x] - I'm updating the code to reflect that now (and I'll re-run the files as well)

bhilbert4, 2018-08-17:
Fixed via #149

closed 2018-08-17 by bhilbert4