spacetelescope/mirage #145: PSF library script could be smarter about choosing SW/LW detectors automatically

mperrin, 2018-08-15:
Right now if you ask for just a short-wave filter, it doesn't know that must mean you only want the short wave detectors. 

```python
>>> mirage.psf.psf_library.CreatePSFLibrary('nircam', filters='F212N')
```
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-b81de98279df> in <module>()
----> 1 mirage.psf.psf_library.CreatePSFLibrary('nircam', filters='F212N')

~/Dropbox (Personal)/Documents/software/git/mirage/mirage/psf/psf_library.py in __init__(self, instrument, filters, detectors, fov_pixels, oversample, num_psfs, save, fileloc, filename, overwrite)
    202                 if "5" in det:
    203                     [self._raise_value_error("short filter", det, filt) for filt in
--> 204                      self.filter_list if filt in CreatePSFLibrary.nrca_short_filters]
    205 
    206                 else:

~/Dropbox (Personal)/Documents/software/git/mirage/mirage/psf/psf_library.py in <listcomp>(.0)
    202                 if "5" in det:
    203                     [self._raise_value_error("short filter", det, filt) for filt in
--> 204                      self.filter_list if filt in CreatePSFLibrary.nrca_short_filters]
    205 
    206                 else:

~/Dropbox (Personal)/Documents/software/git/mirage/mirage/psf/psf_library.py in _raise_value_error(msg_type, det, filt)
    154             message = "You are trying to apply a longwave filter ({}) to a shortwave detector ({}). ".format(filt, det)
    155 
--> 156         raise ValueError(message + "Please change these entries so the filter falls within the detector band.")
    157 
    158     def _set_psf_locations(self, num_psfs):

ValueError: You are trying to apply a shortwave filter (F212N) to a longwave detector (NRCA5). Please change these entries so the filter falls within the detector band.

```

An improved approach might be to test if all the requested filters are short wave, and if so set the detector to short wave: 
```python
if set(self.filter_input).issubset(set(self.nrca_short_filters)):
    self.detector_input='shortwave'
```
etc.     
Alternatively, perhaps the 'all' mode could simply skip over any invalid combinations, without stopping program execution?

mperrin, 2018-08-15:
Some other enhancement suggestions on the PSF creator script. These could be separate issues but for now I'm going to lump 'em all in here. 

 * Right now it looks like the webbpsf function calls are invoking the distortion code (i.e. `add_distortion` is left at its default value `True`), but the PSF used in the library is the undistorted one. The distortion code slows the calculation down for no benefit in that case. I think it would be nice to make this a switch in the PSF lib creator: cases that don't need the distortion details exactly right can turn it off for faster library generation, but if it's turned on then it should be the distorted PSF that gets used in the library. 
 * For the degenerate case of only having a single PSF per detector, that PSF should perhaps be generated in the center (i.e. DET_XY0 = '(1024, 1024)`). Currently it defaults to the corner (0,0). (This is admittedly an arbitrary / aesthetic hand wave - in the case that we're not modeling variation, it's arbitrary where we pick. Alternatively perhaps there could be a way to specify the coordinates for a single PSF. I mention this just because the main use case for having only one PSF per detector in a library is something like OTE commissioning, where we need to make many libraries (because of time variable state) but may only care about one field point per detector. 
 * Some additional keywords could be copied from the WebbPSF headers, in particular for parameters affecting calculation outputs such as the number of wavelengths used, whether the distortion model was applied, etc.

mperrin, 2018-08-15:
What's the benefit of not even allowing any sub-pixel shifts? That will bias all the source photometry by effectively rounding the positions of all sources to the nearest integer pixel. We could certainly do that; I'm just trying to understand better the use case.

KevinVolkSTScI, 2018-08-23:
(Sorry for the late reply, I have been on vacation the past 8 days.)

The purpose of having a single default PSF is to simulate a focus sweep observation for commissioning, with a defocussed PSF from WebbPSF.  In such a case, we are not concerned about getting the stars to the correct sub-pixel position, and we are also not concerned about the photometry bias effect.  Otherwise, there are a few other cases where we may not care about getting the stars to the right sub-pixel positions in simulations and having a single (small) default PSF file would be an advantage as one would not need to make a big PSF library calculation to do such simulations.

The question in the context of this PSF library change is how can I make such a "library" image for the defocussed case?  I have a little WebbPSF driver code to make individual PSFs with some amount of defocussing, but it makes only one PSF at a time and writes out a single two-dimensional file.  I suppose it is not difficult to "re-package" such a file in the format that the new PSF library files have.  

For NIRISS it seems that changing the (x,y) pixel position makes no difference in the output PSF -- at least, I tried this in version 0.7.0 and there was no change in the PSF for the center of the array or the corner of the array.  I might have been doing something wrong, of course.

bhilbert4, 2018-09-13:
Solved by #162. I'll move the discussion about a default PSF to a separate issue.

closed 2018-09-13 by bhilbert4