spacetelescope/mirage #163: Add GUI

bhilbert4, 2018-09-13:
There has been a request to possibly add a GUI for Mirage. Something where you might select an APT file and push buttons to create yaml files and run Mirage. I'm wondering how useful this would be.

Given that we have notebooks that show how to run Mirage from APT file to simulated data, does a GUI add any value or make anything easier?

Also, if you will be simulating all data from an APT file, it may take a long time for Mirage to run. Would you want to have a GUI open that essentially says "Still working" for that long?

mperrin, 2018-09-19:
My impression, speaking as a newbie in using MIRAGE, is that the example notebooks contain a whole lot of "boilerplate" code: lots of setup and configuration and initializing things that would need to be repeated into other notebooks for each case you want to run. I think the intent behind asking for a GUI is making it easier for a new user to get started. It's about easing the learning curve rather than necessarily wanting to have buttons and GUI windows open. 

I think you might accomplish the same goal by continuing to improve the high-level functions that automate most of that, like the `imaging_simulator` functions. You might consider making a new notebook that only shows that approach in a streamlined way, as opposed to the  [current notebook]( https://github.com/spacetelescope/mirage/blob/master/examples/Imaging_simulator_use_examples.ipynb) that shows running all the steps individually before showing the combined approach. 

Could you eventually get to something like the following? 

```python

yfiles = mirage.apt.create_inputs_from_apt(input_xml="somefilename.xml",
                                           pointing_file="somefilename.pointing",
                                            output_dir='/some/path',
                                            use_JWST_pipeline=True,
                                            point_source_catalog='catalog_file.list',
                                            **other_parameters_here)

mirage.run_in_parallel(yfiles)                                  
```

All the complexity of setting up the YAML files and preparing darks and creating seeds  is wrapped up inside that top-level function.  Or perhaps a couple top-level functions to prevent it from needing several hundred different parameters? Not all functionality would necessary be available via this interface. But it would let users get started quickly with reasonable defaults. 

I wrote a GUI for WebbPSF originally, precisely for this reason to let users get started quickly without having to become experts. But I eventually decided that it was better to focus efforts on just the command-line Python programming interface to webbpsf, and make it as easy as possible for users to get reasonable results from that easily by providing good defaults for most parameters, and automating selection of certain options when those made sense.

Johannes-Sahlmann, 2018-09-19:
I completely agree with @mperrin . In my opinion it is a much better use of resources to enhance the command-line interface as opposed to developing a GUI.