spacetelescope/mirage #167: Bug the point source list output.  Also there are issues with sexagestimal input coordinates.

KevinVolkSTScI, 2018-09-27:
I noticed that the DEC position in sexagestimal form has an error for the point sources output file (I do not currently know if this the case also for the galaxies or other object lists).  An example of the output is 

# Field center (degrees):   11.71016667    42.27113333 y axis rotation angle (de
grees): 216.000000  image size: 2048 2048
#
#    Index   RA_(hh:mm:ss)   DEC_(dd:mm:ss)   RA_degrees      DEC_degrees     pi
xel_x   pixel_y    magnitude   counts/sec    counts/frame
1 00:46:50.4400 0+042:16:16.0800    11.71016667    42.27113333  1023.500  1023.500     18.594   4.438777e+03    4.765812e+04

Note the "0+042" in the dec column.  I had found this and fixed it previously, but either I did not put this into the code as a change or it has come back again.  I will find the fix again and submit it.

I also have found that the code does not work for sexagestimal coordinates for the source RA/Dec.  Depending on whether one uses a form like 42:16:16.08 as above or 42d16m16.08s one can get an error in the code before the seed image is generated or one may not, but even when it gets past the initial check with this type of position input it fails later in obs_generator.py in try to float() the string.  I would like to fix this as well.

bhilbert4, 2018-09-27:
You're right in that I believe things are hardwired at the moment to expect the 42:16:34.0 format if a float is not provided. Allowing for the 42d16m35.4s format (and also 5h37m34.6s for RA) would make things more convenient.

Thanks for fixing these.

KevinVolkSTScI, 2018-09-27:
HI,

  It did not take the : versions of the field position when I tried it.  In any case, I will see about fixing these and make sure that all the forms are covered.

Kevin

KevinVolkSTScI, 2018-09-28:
Further investigation has produced the following results:

(1) The issue with RA/Dec positions given with : separators (i.e. 00:46:50.44) is that yaml interprets the values as hh:mm:ss.ss and replaces the string with the equivalent value in seconds.  Hence the above string produces a value of 2810.44.  As there are no remaining signs that the input value has the : symbol in it, there is no way that I can see to adjust the values by dividing by 3600.  One can work around this using quotes around the string.  Hence an input of '00:46:50.44' appears to work properly.
(2) Although sexagestimal strings can be used for object position inputs for the catalogues and they work for the RA and Dec parameters in catalog_seed_input.py, such inputs cause to code to crash in obs_generator.py because it will try to call self.paraeRADec and this routine is in catalog_seed_image.py but not in obs_generator.py.  An obvious solution that fixes the problem is to copy the parseRADec routine to obs_generator.py, although it is not ideal to have the same code in two places.
(3) There is a simple fix to the "0+0" output strings, which is to explicitly replace "0+0" by "+" in the Dec string (one should probably also replace "0-0" with "-" just to be safe).  The zeros come in because the codes replaces leading spaces with zeros in the output string.  However there should be no spaces around the + sign or - sign.  In any case, the fix here is fairly trivial.
(4) Unrelated to the above, I find a divide by zero problem in the part of the code that produces the Sersic profile for galaxies.  A galaxy with a very small radius and a large enough ellipticity apparently produces an initial model image with no signal, whereupon when the code tries to renomalize the signal there is a divide by zero run-time error.  That messes up the seed image by adding a stamp image with all NaN values to the image that is being assembled.  An easy fix is possible, one can just set the scaling factor to zero if the total signal from the stamp image is zero.  The galaxy requested then does not appear in the seed image, but since the stamp image initially is calculated to have zero signal one has no way to properly put in the galaxy anyway.
(4) I also wonder if the "Too small!" message the code produces when the initial stamp image for a galaxy from the Sersic parameters is smaller than needed should be changed.  I would rather see the code expand the stamp image area if it is too small to start with, on the grounds that this part of the calculation is not the part that takes the most time in my testing.  If that is not something people want, at least the message should tell you which parameters in the galaxy list produce the "Too small!" message.

bhilbert4, 2018-09-29:
Thanks for the in-depth investigation. 
1) That's an annoying feature of yaml. I came across something similar a few weeks ago. I can't remember at the moment what it was, but yaml was making assumptions about some input value and changing it, just like it is here. Maybe I'll put together an example source catalog that uses the quotes in order to try and make it clear that they are necessary.

2) I'll fix the parseRADec on Monday. Since it is needed in two places, I agree that moving into `utils` and then importing where needed is the right way to deal with it.

3) The code is replacing leading spaces with zeros? Hmm. So maybe a better fix would be to do `variable_name = variable_name.strip()` before writing it out. This should strip all leading and trailing whitespace. Or, to remove all whitespace (if we want to guard against spaces between the + and the numbers) you can use regular expressions:

    import re
    new_string_variable = re.sub(r"\s+", "", string_variable)

4) That seems like a reasonable solution, along with maybe printing a note to the screen saying that the galaxy at x,y is being skipped because the model returned a total signal of 0. At some point when I get around to adding logging to `mirage`, then this information can be captured in the output log file and reviewed after `mirage` has been run.

5) I definitely want to change this. It seems like there should be a way to determine how large of a stamp is needed to hold the galaxy before actually creating the stamp. I need to play around and see if that's the case though. (See issue #168 for anyone reading this who hasn't seen it yet).

bhilbert4, 2018-10-01:
See PR #175. Do you also want me to take care of the "0+0" issue while I'm making the changes to `parse_ra_dec`? Looking at the code this morning, this turned into a bigger job than I'd anticipated, with the way that the source filtering wasn't working in cases where `parseRADec` had to be called.

KevinVolkSTScI, 2018-10-01:
I can post fixes for the "0+0" problem and the little issue in the Sersic profiles a a separate pull request.  The fixes are fairly simple and should be easy for someone to review.  I just have not got to that yet this week.  Besides the code changes, I should also put in a unit test of the three position forms in the .yam files.

closed 2018-10-09 by KevinVolkSTScI