spacetelescope/mirage #184: Catalog generator functions

bhilbert4, 2018-10-09:
Create some functions for catalog generation.

The motivation behind this was feedback from within the NIRCam team (@marthaboyer) with the idea: 

"Wouldn't it be nice if I could tell Mirage to create a globular cluster catalog rather than having to create one myself."

Catalog creation is one of the more time-intensive things that a potential Mirage user will have to do, so it would be nice to have the ability to create catalogs of representative scenes (stellar fields, extragalactic backgrounds, globular cluster).

Perhaps the first step is a function to create a simple point source catalog following some set of defaults (or user inputs) in terms of luminosity function, source density, etc. for a given pointing. @arminrest suggested that @mcorrenti would be able to provide good default values for stars. We should then think about who to consult as an extra galactic expert in order to get good defaults for those. 

@arminrest also suggested adding the ability for Mirage to ingest catalogs containing magnitudes for multiple filters. Use filter names as column headers, and have Mirage look for those. This would cut down on the number of catalog files needed.

What does everyone think?

laurenmarietta, 2018-10-23:
Sounds like a great - if complicated - feature!

We do have some skeleton of this currently in `catalogs/get_catalog.py`, though that module only queries point sources in specific WISE and 2MASS bands (corresponding to the F212N and F480M NIRCam filters that we use for commissioning). This module would certainly have to be heavily modified to generate catalogs based on fancy parameters beyond just pointing.

bhilbert4, 2018-11-06:
Random thoughts on catalog generator development that came to me as I was trying to get the kids to go to bed:
Create a catalog object, and then options for sources to add to it:
1. Add background stars following some luminosity function (Matteo has example using Besancon in astroquery) with typical densities in/out of plane of Milky Way, in bulge.
2. Add background galaxies following some luminosity function at some density
3. Add globular cluster with N stars and some density profile extending over M arcseconds with a typical luminosity function.
4. Generate a catalog based on real stars using astroquery and 2MASS, Vizier, etc (note that most of these catalogs have a fairly low maximum magnitude, so fainter sources will be missing)

Also have functions to:
1. Take an input SED with real flux units and translate into magnitudes through a specific filter
2. Take a normalized input SED and a magnitude through filter X and calculate the magnitude in filter Y

KevinVolkSTScI, 2018-11-14:
I have various bits of code to do this type of catalogue generation based on 2MASS, WISE and GAIA DR2 data.  There is some cross-matching of positions that is done, although the method is simple and would break down in sufficiently crowded fields because the code does not check that one 2MASS source is not cross-matched to more than one GAIA DR2 source.  All this can be put into mirage, probably fairly easily.  The one caveat is that I am using a code to transform GAIA DR2 blue/red magnitudes to rough NIRISS magnitudes.  One can do this also for NIRCam filters or for the Guiders, but it then requires at least one extra file of simulated magnitudes to transform.

A possible alternative would be to use just 2MASS/WISE photometry and interpolate in wavelength to the NIRISS or NIRCam or Guider pivot wavelength values.  This works for reddened sources, but then is limited to objects with 2MASS and WISE photometry.  I have code for this as well.

Third, I have a code to generate a fake galaxy population based on the UDF catalogue produced by Gabe Brammer.   This catalogue gives AB magnitudes for the 12 NIRISS filters, so it can be adapted for NIRCam easily but for the Guider some additional thought is needed as to what to do.

bhilbert4, 2018-11-14:
That sounds great. Based on our conversation this afternoon, I'm tweaking my catalog generator functions a little bit. 

I'm trying to figure out the best way to collaborate on this. I could push my branch to my fork and then you could add your edits to that branch, and we could eventually merge that into master.

I could also merge my branch into master and then you could create your own branch off of that.

I worry a little bit about the latter since my catalog generation code is not fully functional, although it is close. Maybe I should get things in a good state and then merge that into master? I feel like at that point the bulk of the work is going to fall on you as you add your code and make whatever changes are necessary to get it to work within mirage. Assuming that's true then it probably makes more sense for you to work on your own branch as opposed to both of us working on my branch. What do you think?

KevinVolkSTScI, 2018-11-15:
I think we can work on a branch.  I can make a fork of your branch and work on things.  As long as the merge request comes from my eventual copy of your code then it should work as normal.  So I would hold off merging changes to the master branch until the functionality is in place and working.

bhilbert4, 2018-11-15:
Sounds good. If you branch off of my `catalog_generator` branch and develop in there, then once you're done if I need to make any changes to my functions we can merge your branch back into my `catalog_generator` branch and I can work in there. Then we can merge that into the master branch upstream. We can skip the intermediate step though if I don't need to make any changes after you're done. In either case, I'll hold off on developing any more on my `catalog_generator` branch until you've finished. That should minimize conflicts when it comes time to merge branches.

Just to explain what I've been doing in my branch (in the mirage/catalogs directory):

`catalog_generator.py` - contains code that defines mirage catalog objects (separate objects for each type of catalog), including a function to write out the objects into text files of the appropriate formats so that mirage can use them.

`create_catalogs.py` - contains code for generating catalogs based on queries to various databases (2MASS, GAIA, Besancon). 

I recommend using separate file(s) to contain your photometric transformation code. It may make the most sense to add catalog matching code another new file, or to `create_catalogs.py`. Separating things this way should make it easier to maintain everything later.

astropy has some built-in functionality for catalog matching which may or may not be helpful:
http://docs.astropy.org/en/stable/coordinates/matchsep.html

bhilbert4, 2018-11-20:
2 more details to add:

1. Enable use of the "extinction" keyword for the Besancon query. Value is extinction per kpc in A_V.
2. Look into using the JADES catalog as a means for generating generic extra-galactic background scenes. @NorPirzkal uses this catalog for some of his simulations.
3. Other extra-galactic options: 3D-HST, Candels, Clash

bhilbert4, 2018-11-20:
Brad Whitmore also pointed out how to get source catalogs from the Hubble Source Catalog. Need to check the MAST API to see if it's possible to do that programmatically.

KevinVolkSTScI, 2018-11-20:
I have some code that works with HSC output, but in general I prefer to avoid it in running simulations.  There are a very large number of possible filters which is one issue (as well, the uncertainties in transforming from UV magnitudes to NIRISS/NIRCam/Guider count rates are very large).  What I have seen is that typically one gets a number of HSC sources with slightly different coordinates clustering around a given 2MASS source, WISE source, or GAIA source in a manner that is very difficult to deal with.  Also, the coverage is quite patchy on the sky and locally within a scene.  For all these reasons I suggest not building any HSC capability into mirage.  If that is to be used, the user needs to get the HSC data and figure out how to make some source list that applies to JWST.

KevinVolkSTScI, 2018-11-20:
If one can do a Besancon query and get J and K magnitudes back that is a reasonable input to transform to the NIRISS and NIRCam filters.  I have machinery of that sort using a publicly available star count code named "galaxia" that is similar to the Besancon one.  It is less helpful if the star counts are in UBV because the transformation becomes less accurate.  I think I tried the Besancon query once upon a time but it did not work for me for some reason.

bhilbert4, 2018-11-21:
I played with Besancon queries a few weeks ago. See the functions in `create_catalogs.py` in my `catalog_generator` branch. In my tests, it returned JHKL for all sources (or actually, I think it returned J and then color terms allowing HKL to be easily calculated).

KevinVolkSTScI, 2018-12-10:
I am going to start working on this today and see where I can get to this week.  Apologies on how long it took me to get through the other high priority items that I collected in my visit so I could start on this task.  I am going to try forking your development branch and then see how things work at present, so I may have questions.

KevinVolkSTScI, 2018-12-10:
I have installed the branch and am looking at the functions.  It looks like I need to add a query function for WISE.  The algorithm I am thinking of for getting sources is:

(1) query GAIA DR2
(2) query 2MASS
(3) query WISE
(4) cross match the sources
(5) where sources have GAIA blue/red plus 2MASS plus WISE interpolate the photon weighted mean flux as a function of wavelength (this means that extinction is allowed for in some way)
(6) where sources have GAIA blue/red plus 2MASS used 2MASS K for longer wavelengths and do the same interpolation
(7) where sources have GAIA blue/red plus WISE, convert GAIA blue/red to 2MASS JHK and then do the same interpolation
(8) where sources have only 2MASS, interpolate/extrapolate with the 2MASS JHK
(9) where sources have only WISE, use WISE W1 for the NIRISS/NIRCam/Guider filters at L or shorter, use W2 for 3.75 microns or longer

Step (4) is the tricky one.  If a field is not too crowded a simple cross-matching can work.  Problems may arise if there are many GAIA sources that might correspond to a single 2MASS or WISE source.  The above is similar to what I have now in other codes.

Separate from this, if I can get Besancon model VRIJHKL magnitudes these can be interpolated to the NIRISS/NIRCam/Guider filters in the same general way (assuming extinction effects are included in the Besancon model outputs).

The alternative is to transform from GAIA blue/red or 2MASS to NIRISS/NIRCam/Guider magnitudes.  For Besancon models, one would presumably transform JHKL to NIRISS/NIRCam/Guider magnitudes.  The issue with doing this is that when there is extinction it will fail and it is in general more uncertain for cool stars than for hot ones because of the way A0V magnitudes are defined.  If there is no extinction then it works fine for K-type and earlier stars but may have problems for M-type stars.

Johannes-Sahlmann, 2018-12-10:
Note that the Gaia archive http://gea.esac.esa.int/archive/ contains pre-computed crossmatch tables for many catalogs, including the ones you mention.
A programmatic example of how to use those tables can be found here: https://github.com/Johannes-Sahlmann/scripts/blob/master/gaia_archive/query_with_crossmatch_table.py

KevinVolkSTScI, 2018-12-10:
Good point.  I will use these rather than re-generating my own.

Johannes-Sahlmann, 2018-12-11:
That's a bug in my example script. gaiadr2.gaia_source is independent of gaiadr1.gaia_source . You should update calls to gaiadr1.*_best_neighbour, by gaiadr2.*_best_neighbour.

I will fix that also in my repo.

bhilbert4, 2018-12-13:
Updated requests from users:

Have a switch e.g. 'add_stellar_background' and 'add_galaxy_background' on top of the catalogs supplied by the user. That way the user can specify their objects of interest, and Mirage can take care of creating/adding the background sources to be simulated.  

Mirage may need to make some assumptions in this case. e.g. Stellar background density high/medium/low. 

Also, we've put a lot of work into stellar sources. It would also be nice to work on galactic catalogs. The JADES catalog seems like a good source to use. @NorPirzkal has been using it for his simulations.

KevinVolkSTScI, 2018-12-13:
I have the machinery to do background galaxy simulations based on GOODS-S for NIRISS, and that should adapt for NIRCam fairly easily except for F070W, but for that filter we could just take the magnitude to be the same as F090W for this purpose.  I assume people are not likely to use NIRCam for deep F070W imaging what with WFIRST coming.  For the Guider things are a bit more problematic, but I should be able to make a weighted mean of the NIRISS broad filters values from F090W to F480M that approximates the Guider magnitudes.  So that should be coming soon...

The Besancon code can be used for the "add_stellar_background" command.  It is not too much of a stretch to ask the user to give and RA and Dec value.  They can make something up if they do not have a specific position in mind.  I have that part of things working.  The user would just have to specify the bright limit for the calculation along with the position.

bhilbert4, 2018-12-13:
That sounds promising. I think for GOODS-S (and also for JADES if we decide to use it), there should be an option to treat it like the Besanscon query for stars. In that case sources can be pulled from GOODS-S or JADES and placed at ~random RA, Dec positions in the field of view, if the pointing is somewhere outside the GOODS-S/JADES coverage areas.

The use of Besancon was exactly to try and fulfill this 'add_stellar_background' button request. 

The big question now, which led to quite a debate at the meeting, was whether this catalog generation should live inside of mirage (e.g. run Mirage with an option turned on such that the catalog will be generated by Mirage as it runs, in which case RA and Dec can be grabbed from the input yaml file) versus outside of Mirage, where the user constructs the catalog, and then runs Mirage. Of course the latter case should always be an option. The debate was whether to allow the former case as well.

But that debate at least only really affects where these catalog functions are called, and not the functions themselves.

KevinVolkSTScI, 2018-12-13:
The GOODS-S code right now has no option to actually use the source catalogue if one pointed at the GOODS-S field; instead it makes a random selection of objects from the catalogue, assigns random pixel positions within the field scaled to the NIRISS detector field of view, and makes random changes in the Sersic parameters and magnitudes within some limits compared to the input values so one does not just get the original field for every pointing.  For the Guider the same code can be used once I sort out some reasonable way to combine the magnitudes.  I could change the positions from pixels to sky coordinates and then the same code could be used for an individual NIRCam module whether in short or long wavelengths.  Or the area could be expanded somewhat as long as it remains smaller than the GOODS-S field.  These details need to be worked out, but they should not be too difficult to solve.

bhilbert4, 2019-05-01:
Catalog generation tools have been merged. Closing this. Any new catalog generation requests can be made via a new Issue.

closed 2019-05-01 by bhilbert4