spacetelescope/mirage #190: Fiducial point override use in APT files

bhilbert4, 2018-10-17:
I'm working with a NIRCam APT file that contains an observation where FiducialPointOverride is present (with the NIRCamEngineeringIImaging template). When I ran the `yaml_generator`, I'm not sure it was doing the right thing with the FiducialPointOverride.

In my case, the data were defined as being full frame images in all A module detectors. The FiducialPointOverride was set to NRCA3_DHSPIL, and there were accompanying xoffset and yoffset values. From my reading of the [APT special requirements page on JDox](https://jwst-docs.stsci.edu/display/JPP/JWST+APT+Special+Requirements), this means that the target RA, Dec will be placed at the reference location of NRCA3_DHSPIL. But the way the `apt_inputs` interprets this situation is that it specifies that the data to be created is only that for the NRCA3_DHSPIL aperture (rather than full frame for all A module detectors). It also says that the xoffset, yoffset is not applied to the simulated data, although I believe that it is since those offsets are baked in to the dither values in the `pointing` file.

My thought is that in this case, Mirage should be making exposures for all A module detectors, and that the pointing should be correct since the pointing file contains all the necessary information. @laurenmarietta, does that make sense, or am I misinterpreting how FiducialPointOverride is supposed to work?

laurenmarietta, 2018-10-23:
Yes, I think you are correct. I implemented the FiducialPointOverride part early on in my exploration of `apt_inputs` and I would chalk this up to my own misunderstanding of how that was supposed to work. 

Specifically, I think I ignored the xoffset/yoffset because I didn't know how MIRAGE was using them (you say it's just taken care of in the pointing file?), but these parameters definitely should not be ignored.

I will be happy fix this in a PR ASAP.

In the example you describe, with FiducialPointOverride NRCA3_DHSPIL, would we want to restrict to the module being used (A), or rather to generate yamls for all detectors?

Johannes-Sahlmann, 2018-10-23:
Please wait before implementing this. This part of the code is part of the general revision in #172.
For instance, I already took out the erroneous behaviour related to FiducialPointOverride.

laurenmarietta, 2018-10-23:
Oh! Good to know, thank you. (I am working my way through my GitHub notifications and haven't gotten to that PR yet.)

laurenmarietta, 2018-11-08:
@Johannes-Sahlmann discovered another related problem while testing #172:

> When there is a FiducialPointOverride defined in APT, the corresponding pointing file contains the aperture name of the FiducialPointOverride (because that defines the telescope pointing). mirage gets the aperture_name from that pointing file, and that should be incorrect in most cases when FiducialPointOverride is set. For instance, I get an error with program /01142/OTE08-1142.xml, where an FGS exposure has a special requirement FiducialPointOverride=NRCA3_FULL. Of course, we do not want mirage to generate a NIRCam exposure in that case.
> I wonder how this possibly can be handled correctly in the current master branch? I don't think mirage currently has a way to determine the aperture_name of the underlying exposure?

laurenmarietta, 2018-12-19:
Closed via #214

closed 2018-12-19 by laurenmarietta