spacetelescope/mirage #194: Consistent source of SIAF information

bhilbert4, 2018-10-22:
Currently in Mirage, if a distortion reference file is provided, the model from that file is used for the translation from detector x,y to RA, Dec, and vice versa. However, information such as the V2_ref, V3_ref, pixel scale, and V3SciXAngle are pulled from pysiaf. Further, quantities like the local roll angle and the attitude matrix are derived from these quantities in pysiaf. 

This can lead to problems in the case where the distortion reference file contains slightly different information than pysiaf. For example, the attitude matrix is created using v2_ref, v3_ref. So if these values are different in pysiaf compared to the values used to create the model in the distortion reference file, then the resulting attitude matrix will not be consistent with the distortion model. However, quantities such as v2_ref, v3_ref, pixel scale, and V3SciXAngle are not present at all in distortion reference files. (v2_ref, v3_ref can be pulled out of the model coefficients, but not the other quantities). So how will we get these quantities in the case where we want to use reference files over pysiaf for SIAF information?? **This situation should not arise until after in-flight calibrations are complete and we have, for example, filter dependent distortion models, which pysiaf will not carry.**

Do we assume that pysiaf has the correct values of (e.g.) v2_ref, v3_ref, and that the model in the reference file was built around these values? Do we need to update the format of the reference files so that they carry the SIAF information used to construct the distortion models they contain? Maybe @Johannes-Sahlmann can help clear up my confusion here.

Johannes-Sahlmann, 2018-10-23:
Eventually, it is up to the instrument teams to decide what to put into the SIAF and the WCS reference files. However, the way I currently see it is that SIAF is the determining carrier of V2Ref, V3Ref, and V3IdlYAngle. Filter dependent offsets to these values are carried either in `Filter offset` reference files or absorbed in the low-order coefficients of the WCS reference file. 
The local pixel scale can always be extracted from the polynomial coefficients.
In any case, I will bring this up at the next astrometry collaboration meeting to seek clarification from the SI teams.

pllim, 2019-07-29:
Maybe this should be high priority. I ran into this problem in a round-about way. See spacetelescope/gwcs#241 . Using your simulated image, the N-E compass in Ginga can't display properly because it uses X=0.5 Y=0.5 to calculate the compass display length. Because distortion is bad at this pixel, and this problem, the compass failed.

pllim, 2021-06-18:
I think I saw this again... but I am not sure... Did Mirage make this file?

https://data.science.stsci.edu/redirect/JWST/jwst-data_analysis_tools/stellar_photometry/jw01072001001_01101_00001_nrcb1_cal.fits

bhilbert4, 2021-06-18:
That does look like a Mirage file. There are Mirage-specific keywords towards the bottom of the 0th extension header.

pllim, 2021-06-18:
```python
>>> import jwst
>>> import gwcs
>>> jwst.__version__
'1.2.3'
>>> gwcs.__version__
'0.16.1'
>>> filename = 'jw01072001001_01101_00001_nrcb1_cal.fits'
>>> dm = datamodels.open(filename)
>>> point = (0, 0)
>>> gw = dm['meta']['wcs']
>>> sky = gw.pixel_to_world(*point)
>>> sky
<SkyCoord (ICRS): (ra, dec) in deg
    (80.48173713, -69.49567204)>
>>> point2 = gw.world_to_pixel(sky)
>>> point2  # Expects (0, 0)
[-0.0073084298053345265, -0.021459428156163085]
>>> sky2 = gw.pixel_to_world(*point2)
>>> sky2  # Expects (80.48173713, -69.49567204)
<SkyCoord (ICRS): (ra, dec) in deg
    (80.48173731, -69.49567222)>
>>> sky.separation(sky2).to('arcsec')
<Angle 0.00070436 arcsec>
```