spacetelescope/mirage #197: User input to create updated observation list file

bhilbert4, 2018-10-25:
With the observation list file format being changed as part of #172, we need to think about how users will get the appropriate details into the observation list file. We need some kind of (small) catalog configuration file that can be taken in and have its contents spread correctly into the observation list file produced by `write_observationlist.py`.

Here are some initial thoughts:

Target: myTarget
   Instrument: NIRCam
        Filter: F070W
            Ptsrc_Catalog: mycat.cat
   Instrument: NIRISS
       Filter: F090W
           Ptsrc_catalog: mynirisscat.cat
 
Or reverse catalogs and filters to make it easier to contain multiple filters in a catalog?

Target: myTarget
   Instrument: NIRCam
        Ptsrc_Catalog: mycat.cat
             Filters : (list of filters)
   Instrument: NIRISS
        Ptsrc_catalog: mynirisscat.cat
	     Filters: (list of filters)

But we need the ability to create epochs, and vary PAV3 and background with epochs (if not observations)!

Maybe another section of the file:

Epoch: 1
    Date: 1/2/21
    PAV3: 98
    Background: medium
    Observations: (list of observation names or numbers from APT file)

laurenmarietta, 2018-10-26:
I'm far from sure that this is the best way to do this, but I am finagling this kind of thing, specifically for changing the PSF library path for each exposure, on my `commissioning-psfs` branch.

I implement it in this notebook: https://nbviewer.jupyter.org/github/laurenmarietta/mirage/blob/commissioning-psfs/examples/OTECommissioning_simulator_with_PSFs.ipynb

Basically, I define another attribute of the yaml object: 
```
yam.psf_paths = psf_paths
```
which is then manipulated in `yaml_generator`. Here is the function that formats that input list of PSF paths to distribute throughout all of the yamls: https://github.com/laurenmarietta/mirage/blob/commissioning-psfs/mirage/yaml/yaml_generator.py#L857

If users are defining many things on a observation-by-observation basis, I doubt this is the best way to do this (your idea of including another file seems better for that), but I thought I would share what I had come up with so far.

bhilbert4, 2018-10-26:
So in this case, you have the PSF libraries set up series of appropriately named subdirectories, and that's how you're able to find and associate each one with the relevant exposure. Is that right? Something like that may work for catalogs. We would still need a way to get PAV3 and background values for each observation/epoch/etc. 

I think most important is to come up with a way that is easy for the user. At our training session for NIRCam members, manually editing the observation list file was the top complaint.

bhilbert4, 2018-10-26:
I'm also wondering if we come up with a clever, easy solution, do we really need the observation generation file at all? Can we use this new solution as a replacement for the observation list file?

bhilbert4, 2019-01-17:
Circling back around to this since we have a user who is trying to use galaxy catalogs in his simulations.

I'm thinking that for the moment, the easiest solution would be to add more inputs to (or expand single values into lists within) `get_observation_dict`.  Adding to what exists at the moment for the point source catalogs and the `parameter_defaults`.

Currently for catalogs we use:
catalogs = {'nircam': {'sw': [list of catalogs], 'lw' : [list of catalogs], 
                    'niriss': [list of catalogs]
                   }
In order to make this useful for the other types of catalogs, I propose an update to:
catalogs = {'nircam': {'sw': {ptsrc: [list of catalogs], 'galaxy': [list of catalogs]}
                                     'lw' : {ptsrc: [list of catalogs], 'galaxy': [list of catalogs]}
                    'niriss': {'ptsrc': [list of catalogs], 'galaxy': [list of catalogs]}
                   }
and also update parameter_defaults to accept lists. For example, currently the input can be:
parameter_defaults['PAV3'] = 110.
Switch to:
parameter_defaults['PAV3'] = [110, 110, 20, 20]

I think it probably also makes sense to combine the 'background', 'background_sw', and 'background_lw' into something that looks more like how the catalogs inputs are structured.
backgrounds = {'nircam': {'sw': [low, medium], 'lw': [low, medium]},
                           'niriss': [low, medium]
                           }

Any opinions? @KevinVolkSTScI @Johannes-Sahlmann @laurenmarietta

bhilbert4, 2019-01-17:
Currently, if you produce your yaml files using yaml_generator, a different dark is selected for each exposure. So your latter case of a different dark for each dither position should already be in place (note that we are only picking from a collection of 5 darks though). You're right that ideally there should be a different dark for each integration in the case of an exposure that contains multiple integrations.

KevinVolkSTScI, 2019-01-17:
Perhaps the thing that needs to be done is to have some way to define a larger selection of dark files at the APT conversion step.  One could put more such darks in the $MIRAGE_DATA area, but that is already quite large and I am reluctant to add more large files to the directory.  Only 5 darks to choose from when one has at least 4 dither positions is not enough.  If one has multiple filters and dithering the number of required darks is certainly larger than 5, probably much larger.

bhilbert4, 2019-09-04:
Returning to this. I'd like to get at least the catalog inputs sorted out before making everything fully public. Currently the point source catalogs can be input via a dictionary. Reference file overrides can also be specified via dictionary. So keep to the same strategy for the other quantities?

Quantities that should be varied observation to observation:

- cosmic ray rate? (In the case of a proposal with multiple epochs)
    library and scale would have to vary, observation by observation

- catalogs - There are many different kinds of catalogs. Each should be allowed to vary. Target by target? Observation by observation?

- Background rate - should be allowed to vary observation by observation.  When 'high', 'medium', 'low' are used then it will naturally vary with filter, so it effectively varies exposure by exposure and SW/LW in NIRCam. What about when a number is used? Let's keep things simple at the moment and restrict that to observation by observation as well.

- PAV3 (Telescope:rotation) This should vary observation by observation. I don't think PAV3 is allowed to change on timescales smaller than that. Ideally this should be tied to date-obs/time-obs as well, in order to specify different epochs.

In all cases it would be nice to allow a user to enter a single value if they want to use the same value for all observations/exposures.

For quantities that can vary observation by observation, how would we set up the dictionary for that? Use observation names as keys? What about observations that do not have names in the APT file? Tell users that Mirage requires observations to be named? Another option would be to use the observation number as a key. This would make for shorter keys and observations are, I think, guaranteed to be numbered unless the user deletes the number from the APT file.

Inputs:

**Cosmic Rays**
Single value for all observations:
> cosmic_rays = {'library': 'SUNMAX', 'scale': 1.0}

Observation-dependent:
 >   cosmic_rays = {'001': {'library': 'SUNMAX', 'scale': 1.2},
 >                             '002': {'library': 'SUNMAX', 'scale':0.9}
 >                         }

**Background**
Single value for all observations:
> background = 'high'

Observation-dependent:
> background = {'001': 'high', '002': 'medium', '003': 22.3}

**PAV3**
Single value for all observations:
> pav3 = 34.5

Observation-dependent:
> pav3 = {'001': 34.5, '002': 154.5}

**Catalogs**
Single value for all observations:
> catalogs = {'pointsource': 'ptsrc.cat', 'galaxy': 'gal.cat'}

Target-dependent:
> catalogs = {'my_targ_1': {'pointsource': 'ptsrc1.cat', 'galaxy': 'galaxy1.cat', 'extended': 'ex1.cat'},
>                    'my_targ_2': {'pointsource': 'ptsrc2.cat', 'galaxy': 'galaxy2.cat', 'extended': 'ex2.cat'}
>                   }

Observation-dependent:
> catalogs = {'001': {'pointsource': 'ptsrc1.cat', 'galaxy': 'galaxy1.cat'},
>                    '002': {'pointsource': 'ptsrc2.cat', 'galaxy': 'galaxy2.cat'}
>                   }

**Date-obs**, to go along with PAV3? Dictionary contains the starting date of each observation. Let's not worry about time-obs for now.
Single value.
> dates = '2019-5-25'

Observation-dependent
> dates = {'001': '2019-05-25', '002': '2019-11-15'}

And then add all of these as keyword inputs to the yaml_generator call, with the defaults that are used now if the keywords are not supplied?
@Johannes-Sahlmann what do you think?

mperrin, 2019-09-04:
For PAV3, we certainly need to be able to specify it per observation. There are several possible ways I see to do that, though: 

1. Specify an absolute PAV3, for instance in degrees E of N. This would be independent of date, but might lead to an inconsistency in which MIRAGE is asked to simulate a PAV3 that is not physically realizable. So it might be wise in that case to code up a check for that condition. 
2. Specify a date, plus optionally a delta from the nominal roll on that date.  This could I think more closely mimic how PPS schedules things, which I think is by default placing visits at the nominal roll in the absence of other constraints (we should verify this with the SCOPE gang, though). 
3. (most complicated, perhaps best long term, perhaps impractical): Read in and make use of the PAV3 constraints in APT, including 'same PA as observation X' and so forth. This would reduce duplication of inputs, but might require re-coding a bunch of decision logic to handle all the possible constraint formats.

mperrin, 2019-09-04:
And observation numbers rather than names as the keys. That's a clearly superior option. Observation names are not guaranteed to be unique; observation numbers are.

bhilbert4, 2019-09-05:
My first inclination is to say that Mirage can simulate the data, but it's up to the user to make sure their inputs are realistic. But perhaps at some point we could have Mirage call the TVT to check input PAV3 values and issue a warning if they are not realizable? 

I envision a lot of Mirage users wanting to simulate data before their observations are scheduled, and so they won't have a definite date or PAV3 value to use.  I think in that case it makes sense to have them input their PAV3 guess, rather than guessing a date that corresponds to some PAV3.

I think that eventually making use of the constraints in APT makes sense, but yeah, this would take some work. I was thinking that we could combine the inputs for PAV3 and date, to emphasize that they are related, and with an eye towards future checking via TVT, e.g.

> orientation = {'001': {'PAV3': 21.2, 'date': '2019-12-31'}

But at that point, why not add cosmic rays and background entries into the dictionary as well? That could start to make the inputs a little unwieldy though. And it's then one step removed from having all this info in a yaml/json file, which is the strategy that Mirage originally used, but was poo-pooed by the NIRCam team as too much work.

KevinVolkSTScI, 2019-09-05:
For whatever it is worth, I think it is much better to leave it to the user to simulate scenes at arbitrary dates and orientations.  It is easy enough to look at a program in the APT and see a possible date of observation and associated range of PAV3 values.  The date is not important for the simulation, so I would say that spending time trying to get that "right" is not useful.  People can always adjust the PAV3 value by hand if they want to after the .yaml files are generated, and unless an observation has a large number of individual exposures at different times one would generally expect that the PAV3 value will not rotate much in a given set of exposures.