spacetelescope/mirage #198: Collect APT Programs for Testing

laurenmarietta, 2018-10-29:
I have put some OTE APT programs (the pointing and xml files) in this directory for us to use to test MIRAGE:

`/ifs/jwst/wit/mirage_data/test_programs`

Other test programs of interest should also be added here, so they are easily located in a shared space.

Johannes-Sahlmann, 2018-11-05:
Thanks for posting the programs. Looks like `01069/NRC21_pid1069_2018_rev2.xml` is inconsistent, because it specifies `<CoordinatedParallel>true</CoordinatedParallel>` but it does not actually contain parallel observations.

bhilbert4, 2018-11-06:
Oops. 1069 might have been my stripped down version of another APT file so that I could make yaml files. I'll try to dig up the original version.

laurenmarietta, 2018-12-19:
I think I'll close this issue now, since we did collect more APT programs to integrate into the test suite.

closed 2018-12-19 by laurenmarietta