spacetelescope/mirage #200: Implement code coverage reporter

laurenmarietta, 2018-11-05:
To help with writing and maintaining effective tests, it might be useful to integrate a code coverage reporter into the MIRAGE CI. I know WebbPSF currently uses [Coveralls](coveralls.io), while JWQL just implemented [codecov](codecov.io).

Either one would be free for us to use with MIRAGE, since the repo is public.

I am curious to hear what others think of this idea?

bhilbert4, 2018-11-05:
Other than embarrassment at the low percentage it's going to find, I think it's a good idea. :) It may be a bit of a challenge to come up with tests for functionality that requires a lot of memory, but I think we'd be able to work around most of the trouble.

Johannes-Sahlmann, 2018-11-05:
Good idea. I tried (not very hard) to implement it for pysiaf, but did not get it to work.