spacetelescope/mirage #210: Set up mirror repository on grit

laurenmarietta, 2018-11-09:
I am talking with @kjbrooks about integrating MIRAGE into the MAGIC tool, for simulating FGS images from OTE commissioning. In order for that to work out, it would be ideal if there could be a mirror repo of MIRAGE on grit.stsci.edu, so that the package will be accessible from the SOGS network.

I believe ST just updated our GitLab software to be the Enterprise edition, so setting up a mirror should be pretty easy.

If everyone thinks this is an okay idea, I would be happy to make it happen!

Johannes-Sahlmann, 2018-11-09:
Sounds good. I think this is passive on the github side, so maybe no change is necessary here?

laurenmarietta, 2018-11-13:
It does look pretty simple, and all happens on grit. Here are the directions for setting up a pull repository with GitLab: https://docs.gitlab.com/ee/workflow/repository_mirroring.html#pulling-from-a-remote-repository-starter

My instinct is that @bhilbert4 should be the owner of the grit repo - if you agree, would you mind setting it up sometime? No rush!

bhilbert4, 2018-11-13:
Sounds good! I'll look into it.

bhilbert4, 2018-11-13:
Should I set it up under my name? There is currently no `JWST` group. I could create a new group ('jwst-simulations'?), although it's not clear if that would be helpful or not.

laurenmarietta, 2018-11-14:
Hmm, I really don't have an intuition for how grit is used by most people... The project I have on grit is under the WFSC group, though, so maybe making a new group is the way to go? I'm not sure if there is anyone to ask who would be more knowledgeable about this, either.

bhilbert4, 2018-11-14:
I've copied mirage over to grit, into a `jwst-simualtions` group. Mirroring isn't yet set up. When I tried this afternoon, it only allowed me to select `push` mirroring, but what we want is `pull` mirroring. I still need to figure out why the drop-down menu with those options wasn't active.

laurenmarietta, 2018-11-14:
Huh, that's odd. I saw on the it_tech_talk Slack channel that ST upgraded GitLab to the Enterprise Edition recently, which as far as I understand should have made pull mirroring an option...

bhilbert4, 2018-11-16:
Whatever the issue was the other day, it seems to be fixed. Mirroring has been turned on and appears to be working successfully.
https://grit.stsci.edu/jwst-simulators/mirage

closed 2018-11-16 by bhilbert4