spacetelescope/mirage #225: Update yaml_generator to correctly deal with WFSS observations

bhilbert4, 2018-11-20:
Before the recent yaml_generator update, the code dealt with WFSS observations in a non-optimal way. It generated appropriate yaml input files, but these had to be manipulated by the user before creating simulations. After this latest update, I find that the generator is no longer producing the correct number of input yaml files. 

This needs to be updated so that WFSS observations result in the correct number of yaml files, and in a way that required little user intervention.

Keep #220 in mind when making this fix, as well as #224.

bhilbert4, 2018-12-04:
@KevinVolkSTScI, I'm working on getting the yaml_generator to work properly with NIRISS WFSS observations. I have it working for the case with a stand alone WFSS observation, but I'm having trouble with the cases where it is parallel or where it is prime and there is another instrument as parallel. 

I'm basically reverse engineering what APT shows me for each type of observation, and I'm wondering if you have some experience with this and can answer my questions.

For the case where NIRISS WFSS is parallel, I select the primary and subpixel dither types and numbers on the tab for the prime instrument. There is no option to specify this on the NIRISS WFSS tab, as expected. In my test case, I have NIRCam imaging as prime, and I chose 3 primary dithers and a subpixel pattern of 2-POINT-LARGE-WITH-NIRISS. I also set `Direct image exposures` to NO_DITHERING. So I expect to see 6 total dithers for all observations. But when I export the xml from APT, it specifies 4 primary dithers for NIRISS with a MEDIUM dither size. I have no idea where this info is coming from, and it seems to conflict with the dithers specified on the NIRCam tab. I don't see any note about it on the JDox page I was looking at. Am I misinterpreting the requirements when these two modes are used in parallel?

KevinVolkSTScI, 2018-12-04:
@bhilbert4 you are correct the case where NIRISS is in parallel for WFSS has dithers from the primary program and should not be getting the values from the NIRISS side.  As far as I know each of the NIRISS WFSS individual exposures (i.e. the direct image and then the grism images) are supposed to happen one per primary program dither position, whatever those may be.  I can check with Paul about the details if you wish.

bhilbert4, 2018-12-04:
I'm definitely curious to know what's going on. I was considering putting in a help desk ticket to the APT team, but confirmation from Paul that something isn't right would be helpful.

bhilbert4, 2018-12-11:
These updates are (I believe) complete in #233. Once I get the new associated tests finalized, the PR will be ready for review.

bhilbert4, 2019-08-08:
PR #233 merged. WFSS is now supported in the yaml_generator.

closed 2019-08-08 by bhilbert4