spacetelescope/mirage #229: Conflict between mkl and numpy in environment_osx.yml

laurenmarietta, 2018-12-04:
@jaymedina was trying to install the MIRAGE environment today using the `conda env create -f mirage_environment_osx.yml` command, and got the following error:

```
The following specifications were found to be in conflict:
  - mkl==2019.0=118
  - numpy-base==1.14.3=py36h479e554_1 -> mkl[version='>=2018.0.2,<2019.0a0']
```

I then tried to create a new environment and I got a similar, but different, error:

```
The following specifications were found to be in conflict:
  - mkl==2019.0=118
  - mkl_fft==1.0.6=py36hb8a8100_0 -> mkl[version='>=2018.0.3,<2019.0a0']
```

I'm a little surprised that this issue hasn't come up before? But it seems like the `mkl` package, whatever that is, is being specified to too high of a version in the environment yaml?

laurenmarietta, 2018-12-04:
(Also, this problem makes me wonder if we should test this file in our Travis pipeline somehow.)

Johannes-Sahlmann, 2018-12-04:
This problem has also been reported by @jotaylor and should be solved asap. 

I somebody already looking deeper into this?

Johannes-Sahlmann, 2018-12-04:
It looks like the issue can be mitigated by replacing 
```
  - mkl=2019.0=118
  - mkl_fft=1.0.6=py36hb8a8100_0
  - mkl_random=1.0.1=py36h5d10147_1
```
by
```
- mkl=2018.0.3=1
- mkl_fft=1.0.4=py36h5d10147_1
- mkl_random=1.0.1=py36h5d10147_1
```
in the `mirage/mirage_environment_osx.yml` file.

Then `conda env create -f mirage_environment_osx.yml python=3.5` created the environment successfully on my local machine.

The origin of the problem, however, is obscure to me since the switch to mkl=2019 was made already at the end of September 2018.

bhilbert4, 2018-12-04:
The last time I checked (which was a while ago), the `jwst` package had a conflict with `numpy-base 1.15`, which is why that is pegged to 1.14 in the environment file. 

Have you looked at the linux environment file? Should we make the same changes in there?

Johannes-Sahlmann, 2018-12-04:
I did not try the linux environment file. (the problem was reported for osx)

jotaylor, 2018-12-04:
I tested it on the plhstins servers and besides a separate installation issue (see #231), the same packages do need to be updated to their linux-specific counterparts. This worked for me:
``` 
  - mkl=2018.0.3=1
  - mkl_fft=1.0.4=py36h4414c95_1
  - mkl_random=1.0.1=py36h4414c95_1
```

jaymedina, 2018-12-05:
instead of creating a separate environment just for MIRAGE, is there any opposition to just cloning the repository and running the `python setup.py install` command, if all the dependency requirements are met? I am more interested in running MIRAGE in a pre-existing environment.

Johannes-Sahlmann, 2018-12-05:
@jaymedina from my point of view, there is no opposition to that. I am running mirage in that same way.

commit b3ca192f (2018-12-05): Merge pull request #230 from Johannes-Sahlmann/fix-mkl-dependency

closes #229
closed 2018-12-05 by Johannes-Sahlmann