spacetelescope/mirage #231: RHEL setup.py installation fails

jotaylor, 2018-12-04:
The updates to use `relic` introduced in a75002c76ae043ac178910eab1598f1b5ed69219 cause `python setup.py install` to fail on STScI's RHEL servers. The issue comes down to the git version ST's servers/RHEL supports (1.7.1 from 2010!!!!!). `relic.git` tries to invoke the `--count` option for `git rev-list` which does not exist in git 1.7.1. 

There are 2 workarounds- individual users update their git versions by compiling from a newer tarball source (may be intimidating for some users) or remove `relic` references.

FWIW, I don't know how many internal users use mirage on the linux servers. It could be the only person this affects is me... Right now I'm using an updated local build of git to get around this issue.

bhilbert4, 2018-12-05:
After several requests, we finally got IT to update the version of git on the witserv machines a few weeks ago. They upgraded from 1.7.1 to 2.19. In that case "module load git" on the command line activates the new version of git. (As I check for a second time in a new window, I see that that command doesn't stick. Apparently you have to use it each time you open a new terminal window!) Maybe IT can also update git on `plhstins` (although maybe without the need to issue a command each time to use it). 

I don't think any Mirage users other than Kevin are working on a linux machine at the moment, but it'd be good to get the linux installation working better. Annoying that the only solution we can make directly is to remove a feature of the package.

bhilbert4, 2018-12-19:
I have a [ticket](https://stsci.service-now.com/stars?id=ticket&table=sc_req_item&sys_id=b0221301dbf06bc042685434ce961993) open with IT asking them to make git 2.19 the default on the witserv machines, and asking about other linux machines. Maybe we need to inundate IT with multiple requests for git to be updated? It seems ridiculous to me that 8 or 9 year old versions of git are driving software development. Is there any other way we can get IT to pay attention to the issue?