spacetelescope/mirage #238: Variable PSF stamp sizes based on brightness

bhilbert4, 2018-12-13:
For very bright sources, the edges of the PSF stamp are visible in the output image. In those cases we ideally should be using larger stamp images. 

Armin's suggestion: 
Multiple versions of library with different stamp sizes, and Mirage picks from the library depending on source brightness. 
Or one library with large stamps, and crop according to source brightness. 

But how does this work with the upcoming new subsampled PSF library?

laurenmarietta, 2018-12-14:
I know @shanosborne and @mperrin put thought into this when Shannon started writing the PSF library generation code. (Tagging them while knowing that Shannon just left for vacation and might not see this for a few weeks.)

I have these notes from a discussion on August 3rd:
* PSF size (big files slow down interpolation time)
    * Create large PSFs in the library, could include case in the library query to crop the PSF if necessary
    * Could imagine a threshold where you calculate where to crop based on the background noise rate (maybe have some sort of idea for each detector what the average dark per exposure time is?)
    * Look into the maximum size PSF for each filter/detector combination (how??), put that into the library, then crop within the code before the interpolation happens
    * Don’t solve the whole problem
    * Come up with a reasonable limiting point, somehow - pretty but not extremely bright star will need x sized PSF, keep this as a parameter that can be changed over time

So, it seems like we were talking about just making all the PSF library files large – Armin's second idea – and then cropping them down as needed. Not sure if that strategy still seems best, or how difficult it would be to implement.

bhilbert4, 2018-12-14:
That all sounds very similar to what we discussed yesterday. Armin suggested dividing up into a handful of magnitude bins. For sources with magnitudes between m1 and m2, crop the PSF to one size. Crop to another size for magnitudes between m2 and m3.

I can envision a case where there are some number of magnitude bins and the sizes to crop to are determined using your second bullet point above.

bhilbert4, 2019-08-08:
This has been implemented in #134

closed 2019-08-08 by bhilbert4