spacetelescope/mirage #239: needed for AMI simulationsuse not yet implemented

anand0xff, 2018-12-17:
this is because the APT interface functionality does not yet support the NirissAmi template.
ValueError: No protocol written to read NirissAmi template.

I cannot drop the appropriate .xml & .pointing files successfully -I zip them below:
[com1093files.zip](https://github.com/spacetelescope/mirage/files/2686944/com1093files.zip)

Johannes-Sahlmann, 2018-12-17:
Hi @anand0xff 
PR #240 introduces the necessary changes and works for me so far. Can you please try this branch out?

anand0xff, 2018-12-18:
Hi Johannes
I cloned your mirage and set git branch to the ami one, and ran the
following (the yaml creation worked, then I switched it off and added the
mirage simulator).