spacetelescope/mirage #251: Inputs from the APT file can cause Mirage to crash

KevinVolkSTScI, 2019-01-16:


KevinVolkSTScI, 2019-01-17:
I looked online and I see that sys.exit and exit both do much the same thing.  Nonetheless, if an error is raised in the code that will not cause the end of a Python session, so I suggest that the code raise an error condition rather than exiting.

bhilbert4, 2019-02-11:
You must have been running with a manually created/edited yaml file? Looking at the code in `yaml_generator`, there is already a check for colons in the proposal title. If they are there, they are replaced with a comma prior to writing out the yaml file.

In terms of avoiding this issue when reading in a manually-created yaml file, the only option I can think of would be a function that reads in the yaml file line by line and replaces/removes any colon after the first, before then moving on to loading the contents of the file into a yaml dictionary.

As for the sys.exit, I'll replace that with a properly raised exception.

KevinVolkSTScI, 2019-02-11:
@bhilbert4 I believe that the cases where this has happened were all automatically generated and the offending line in the .yaml files were not changed by me after the generation.  I did go in and fix the dark ramp names because these were not being handled properly, and I might have also changed some of the other parameters used by the code, but not the title which just goes into the header.  It may be that the fix was put in after I had the issue last month.

bhilbert4, 2019-02-12:
I made some changes in #259 to address this.

bhilbert4, 2019-02-12:
Solved via #259

closed 2019-02-12 by bhilbert4