spacetelescope/mirage #267: Cosmic ray probability is scaled inversely to the frame time

KevinVolkSTScI, 2019-03-01:
I tried running a NIRISS model for the SUB64 subarray (64x64 pixels).  I noticed that the cosmic ray probability per pixel had a much higher value than expected.  From the log of the model I saw

Frametime is 0.05016
Base cosmic ray probability per pixel per second: 0.0012266746411483253

Looking in the code the obs_generator.py file has the following (on line 1795 in the version I am looking at

        self.crrate = self.crrate/self.frametime

I am certain that the rate need to be multiplied by the frametime, not divided by it.  Unless I am mistaken somehow, this is an easy thing to fix.  On the other hand the simulated cosmic rays in all the full frame models that have been made are at too low a rate.  I doubt that this is a real problem given the uncertainties in the cosmic ray rates from the Robberto report which these are based on.  @bhilbert4 do you agree that the frame time should be multiplied by the rate?

bhilbert4, 2019-03-02:
I think you're right, unless I somehow start with the cosmic ray rate per frame and then line 1795 is converting to the rate per second, which seems unlikely. I can have a look at the code over the weekend or Monday to confirm.

In general, if you want to increase the cosmic ray rate, you can always increase the `cosmicRay:scale` factor in the yaml file, or switch to a different CR library, like `SUNMAX or FLARE`.

KevinVolkSTScI, 2019-03-02:
The rate values in the code for the different options (i.e. FLARE) appear to be some type of per second values.  I would have to look back at the document to verify, but that is what I remember was in my code originally.

PR #268 (2019-03-04): Fix CR rate: multiply by frametime
This PR fixes a typo in determining the cosmic ray rate. Previously the rate (CR/pixel/second) was being divided by the single frame exposure time. This changes it so that Mirage is multiplying by the frame time in order to arrive at the CR rate per pixel per frame.

Closes #267
closed 2019-03-04 by bhilbert4