spacetelescope/mirage #274: Problem when using NIRISS .asdf distortion reference file for AMI

Johannes-Sahlmann, 2019-03-07:
It appears that there is a problem when using the .asdf distortion reference file for AMI simulations, in the sense that the sources do not appear in the expected location. When mirage uses pysiaf instead, the source locations are as expected.

Example code that can exhibit the problem is available on a development branch:
https://github.com/Johannes-Sahlmann/mirage/blob/sahlmann-ami-tests/examples/niriss_ami_example/niriss_ami_example.py

bhilbert4, 2019-03-07:
I noticed a similar situation last week when trying to make NIRCam TA subarray data. Pysiaf gave the correct position, while mirage did not. I thought at the time that it was a problem with the reference files, but since you're seeing it as well it seems like there must be a bug somewhere in Mirage.

Johannes-Sahlmann, 2019-03-07:
Good to know.