spacetelescope/mirage #276: Mirage packages output wrongly for sub-array models, crashes the pipeline

KevinVolkSTScI, 2019-03-18:
I have run a mixture of full-frame and sub-array NIRISS models in Mirage for NIRISS CAR 1094 (photometric calibration).  When trying to run the output _uncal.fits files through the pipeline I find that the full-frame models work fine but the sub-array model that I tried fails.  Looking at the structure of the output _uncal.fits files I see problems for the sub-array models.

For the full-frame case the structure of the _uncal.fits file is along the following lines:

witserv3 car1094_simulations > fitsinfo jw01094003002_01166_00038_nis_uncal.fits 
Filename: jw01094003002_01166_00038_nis_uncal.fits
No.    Name      Ver    Type      Cards   Dimensions   Format
  0  PRIMARY       1 PrimaryHDU      87   ()      
  1  SCI           1 ImageHDU        48   (2048, 2048, 30, 1)   int16 (rescales to uint16)   
  2  ZEROFRAME     1 ImageHDU        12   (2048, 2048, 1)   int16 (rescales to uint16)   
  3  GROUP         1 BinTableHDU     36   30R x 13C   [I, I, I, J, I, 26A, I, I, I, I, 36A, D, D]   
  4  ASDF          1 BinTableHDU     11   1R x 1C   [3592B]   

while for the sub-array model the structure of the _uncal.fits file that is produced looks different:

witserv3 car1094_simulations > fitsinfo jw01094001001_01101_00001_nis_uncal.fits
Filename: jw01094001001_01101_00001_nis_uncal.fits
No.    Name      Ver    Type      Cards   Dimensions   Format
  0  PRIMARY       1 PrimaryHDU      87   ()      
  1  SCI           1 ImageHDU        40   (10, 64, 64, 200)   float32   
  2  PIXELDQ       1 ImageHDU        11   (10, 64)   int32 (rescales to uint32)   
  3  GROUPDQ       1 ImageHDU        11   (10, 64, 64, 200)   uint8   
  4  ERR           1 ImageHDU        11   (10, 64, 64, 200)   float32   
  5  ZEROFRAME     1 ImageHDU        10   (10, 64, 200)   float32   
  6  GROUP         1 BinTableHDU     36   12800R x 13C   [I, I, I, J, I, 26A, I, I, I, I, 36A, D, D]   
  7  ASDF          1 BinTableHDU     11   1R x 1C   [3872B]   

The item that causes the JWST pipeline to crash is the PIXELDQ extension, which should be 64x64 pixels but comes out 10x64 pixels.  This must be an error in the Mirage code.  The zero frame extension also has the wrong dimensions, but for both cases I do not see why NIRISS output have this extension in the first place.

The simplest fix is not to have a PIXELDQ array in the output in the first place; I do not know why it is there for sub-arrays but not for full-frame models.

KevinVolkSTScI, 2019-03-18:
I know that NIRCam has zeroframe data for all the ramps.  NIRISS does not have such and it will not be in the DMS data files.  This extension should not be generated for NIRISS or Guider simulations.  I assume that this will simply be ignored by the pipeline, but note that the zero frame extension dimensions are wrong for the NIRISS sub-array model.  If it is retained, this should be fixed as well.

bhilbert4, 2019-03-18:
Interesting. This implies that the `Level1bModel` is not being called for subarray data, as that should automatically scrub extensions other than `SCI` and `Zeroframe`.  I'll have to run some tests to see what's going on.

KevinVolkSTScI, 2019-03-18:
I did try to strip out the zeroframe extension and fix the DQ array from one of the Mirage output _uncal.fits files for the pipeline.  For some reason the pipeline requires a zeroframe extension.  I have not seen this before when using CV3 data in the pipeline.  There is something going in the pipeline that I do not understand.

bhilbert4, 2019-03-18:
Are you working on the master branch of the repo? When I try making a raw subarray file, I see:

```
In [3]: h.info()                                                                                                                                                                                                         
Filename: outputs/nircam_imaging_sub400p_nrcb1_uncal.fits
No.    Name      Ver    Type      Cards   Dimensions   Format
  0  PRIMARY       1 PrimaryHDU      89   ()      
  1  SCI           1 ImageHDU        48   (400, 400, 10, 1)   int16 (rescales to uint16)   
  2  ZEROFRAME     1 ImageHDU        12   (400, 400, 1)   int16 (rescales to uint16)   
  3  GROUP         1 BinTableHDU     36   10R x 13C   [I, I, I, J, I, 26A, I, I, I, I, 36A, D, D]   
  4  ASDF          1 BinTableHDU     11   1R x 1C   [3616B]   
```

bhilbert4, 2019-03-18:
It does look to me like the Level1B datamodel in the jwst package requires a zeroframe extension. I didn't realize that NIRISS wouldn't be carrying that extension. It might be worth opening an issue in the (jwst)[https://github.com/spacetelescope/jwst] repo. I just searched their issues and I don't see any relating to zeroframe being instrument-specific.

KevinVolkSTScI, 2019-03-18:
I will try with the current version of Mirage.  I have to look at some of the converted CV3 data that I have run through the pipeline to see if there is some type of fake zeroframe extension.  NIRISS has no mechanism for storing the first frame as NIRCam does.  It would be nice to have when using the NIRISS NIS read-out mode, but we do not get that data as far as I have ever been told.

KevinVolkSTScI, 2019-03-19:
The issue was that the call to save_DMS was not made with mod='1b' in the first try.  Once that was corrected, the output is as expected.

closed 2019-03-19 by KevinVolkSTScI