spacetelescope/mirage #286: Gridded PSF library creation details

bhilbert4, 2019-04-11:
### Purpose
PR #134 introduces code to use gridded PSFs (in photutils' griddedPSFModels) rather than the PSFs libraries currently in use. 

The purpose of this issue is to discuss details of the gridded PSF libraries to be created for Mirage. We want these libraries to result in simulated data that meets the needs of Mirage users, while trying to minimize the total size (in terms of disk space needed) of the files, as users external to the Institute will need to download them.

### Test Case
For code development, I worked with an example NIRCam gridded PSF file with the following parameters:
filter: F200W
301x301 pixel field of view (matches those in the current libraries)
oversampled by a factor of 5
16 psfs across the detector
"requirements" OPD
default number of wavelengths per PSF (3 for wide filter)

For comparison, the current Mirage PSF library for NIRCam is at a resolution of 0.25 pixels (e.g. there are PSF files for a grid of (0, 0.25, 0.5, 0.75 pixels) in x and y directions.

From my testing while writing the code, I found that the PSFs from the gridded library above were placed onto the detector in the simulated images more accurately (relative to the coordinates specified in the input source catalogs) by about an order of magnitude compared to those from the current library. This was determined by creating the simulated seed image (which is noiseless) and then using `photutils` to locate and calculate the centroid for the sources in the data. Working with a (**very limited**) set of point sources, I found that when using the current library, the worst case had the sources centered ~0.2 pixels from the requested locations. When using the gridded library, the worst case sources were only ~0.02-0.03 pixels from the requested locations. In the NIRCam SW channel this corresponds to just under 1 mas. 

### Astrometric Performance Goal
The most stringent demand on source locations for NIRCam comes from the case of creating simulated data from which source catalogs will be made and fed to NIRSpec for follow up observations. In that case sources need to be known (i.e. placed by Mirage) accurately enough to fall within an MSA shutter. (An MSA shutter is 0.20" × 0.46") This implies, I think, that oversampling by a factor of 5 is sufficient for F200W. This value will change with wavelength though.

### Photometric Performace
Here are the webbpsf notes on subsampling required to be able to measure source flux at a given distance with a given SNR:
https://webbpsf.readthedocs.io/en/stable/sampling.html

From this, it looks like the gridded library I used should be sufficient to measure flux with a SNR of 20 out to 1" from the source. **Is this good enough for most users?**

### Run-time Performance
Note that creating PSFs using more wavelengths will take longer, but will not increase the file size of the library. So there is no cost to the Mirage user to have gridded PSF libraries that come from more wavelengths. Increasing the subsampling will come at a cost, as Mirage's computation time will increase due to the larger array size. The files will also be larger when using more finely sampled PSFs.

I'd like to hear others' thoughts on these libraries, or on what kind of performance they are expecting from Mirage.

@mperrin @KevinVolkSTScI @Johannes-Sahlmann @arminrest @leonardoubeda @mcorrenti

mperrin, 2019-04-11:
Re: photometric performance, the gridded psf library implementation contains a step in which we convolve by a model of the detector pixel response function. That step wasn't included in the calculations of photometric SNR in that webbpsf page. Given that I expect the gridded PSF library should perform even better than what you have stated there. In theory anything Nyquist sampled plus the pixel response function should be quite accurate.

KevinVolkSTScI, 2019-04-11:
For my purposes I am usually concerned with overall photometric accuracy including (in some cases) the far wings of the PSF.  For cases where this is critical I would just use a 9x9 oversampled PSF, rather than the "normal" 5x5 oversampling.  Especially for the Guiders more detailed wavelength sampling than the default in WebbPSF is needed, and one needs to run PSFs for different input spectral shapes as these produce rather different output PSF images.  For general users the S/N of 20 at 1 arc-second should be more than enough.
The last time I tried to see if WebbPSF changes the NIRISS PSF with position on the detector, in the current WebbPSF version, that functionality was not working.  Does that affect this type of gridded PSF library?
Given the undersamping of some of the NIRISS filters I am not sure that 0.02 to 0.03 pixel "placement" accuracy will be obtained, but we need to test this.

bhilbert4, 2019-04-12:
NIRCam also has commissioning and calibration activities where we want to look in detail far out into the wings of the PSF, and we do want to simulate those data before commissioning. For that case, my plan was to make a separate PSF library from that which will be distributed with Mirage. That library would contain much larger PSFs (in terms of area). Similar to the way Mirage currently chooses the appropriate PSF file from the library, with the gridded PSF libraries, the user simply provides the path to the library files and Mirage chooses the file based on instrument/filter/pupil being simulated.

We'll have to think about the best way to select library files for the Guider if the target spectral shape is going to be another selection criteria.

One of the major selling points of the griddedPSFModel is that it captures the distortion effects across the detector. The grid contains subsampled PSFs at various locations across the detector (including distortion effects). Then when Mirage asks for a PSF at pixel (x,y) the griddedPSFModel interpolates between the PSFs in the grid to create the PSF at (x,y).

Here is the webbpsf documentation showing a NIRCam grid:
https://webbpsf.readthedocs.io/en/stable/psf_grids.html

Johannes-Sahlmann, 2019-04-12:
I am looking forward to trying out this new gridded library. The improvements related to source placement accuracy should be immediately evident in the simulations for geometric distortion calibrations that I am performing for FGS, NIRISS, NIRCam. 
I don't any specific requirements or suggestions to add at this time, but can you please explain how this will be deployed? Will there be a switch available to choose between the traditional and new libraries?

bhilbert4, 2019-04-17:
@Johannes-Sahlmann the changes in the code needed to implement the new libraries were fairly invasive. I think this is going to be a new version of Mirage that is not backward compatible with the old libraries. From a user's perspective, everything (entries in the yaml inputs, etc) will remain the same.

bhilbert4, 2019-04-17:
From discussions with @mperrin about the data used by webbpsf to create interpolated PSFs, it seems that Mirage's gridded PSF libraries should contain 3x3 grids of PSFs for NIRCam SW, NIRISS, and FGS. NIRCam LW can support a larger grid. 4x4 or 5x5 perhaps.

KevinVolkSTScI, 2019-04-17:
When I tried the position-dependent PSF function in WebbPSF for Mirage I could not see any change in the output PSF no matter what pixel coordinates were entered.  Of course I may have been doing something wrong but this needs to be looked into a bit before the library is generated.

bhilbert4, 2019-04-17:
I just tried a test case for NIRISS and I see the PSF vary with position:

```
import webbpsf
import copy
import matplotlib.pyplot as plt

# Make 3x3 grid of NIRISS PSFs
nis = webbpsf.NIRISS() 
nis.filter='F200W' 
grid = nis.psf_grid(num_psfs=9, all_detectors=False)

# Display the PSFs and the differences between the PSFs and the mean PSF
webbpsf.gridded_library.display_psf_grid(grid)
plt.show()
```

KevinVolkSTScI, 2019-04-17:
OK, that is good.  I was not doing it this way (I was specifying pixel corrdinates in calls to the PSF generation routint) but as long as it works this way I can use it as needed.  Thanks Bryan.

mperrin, 2019-04-17:
Setting pixel coordinates directly also works fine:
```
ns = webbpsf.NIRISS()
ns.filter="F480M"

ns.detector_position = (200,200)
p0 = ns.calc_psf()
ns.detector_position = (1024,200)
p1 = ns.calc_psf()
webbpsf.display_psf_difference(p0, p1, title= "PSF difference", 
                               ext1=1, ext2=1, imagecrop=2)
``` 
[image]

Kevin, can you provide code examples for what you were using? I think you're right that you must have been doing something wrong when you tried it.

mperrin, 2019-04-17:
Note, there are some open PRs against WebbPSF (in particular https://github.com/spacetelescope/webbpsf/pull/283 by @jarronL) which improves the functionality of the PSF grid generation. This is related to the ISIM field points being near but not actually at the corners of the detectors - while the grid points are at the corners, therefore outside of the ISIM test coverage area. 

@bhilbert4 you should wait until after we have merged that PR 284 into WebbPSF before generating the PSF grids for production use in MIRAGE. Of course you can go ahead and develop & test with the current code for right now. I should have time to look at and merge that PR hopefully next week (last couple weeks have been crazy for me with travel, JWST ops rehearsal, & more travel).

bhilbert4, 2019-09-04:
Closing this issue since gridded libraries have been implemented, and updated versions have been made now that https://github.com/spacetelescope/webbpsf/pull/283 has been merged. Any further issues with the libraries can be dealt with in new issues/PRs.

closed 2019-09-04 by bhilbert4