spacetelescope/mirage #299: support for NirspecMOS and MiriImaging template

stacchella, 2019-05-22:
Hi, 

as part of the NIRCam GTO, we do our imaging as parallels with NIRSpec and also with MIRI. Currently the APT xml reader does not support those templates. As an example, I've included one xml file that uses the NirspecMOS (and MiriMRS) template. Would it be possible to add those APT templates? Thank you for your consideration.

[1180_pa_41.xml.zip](https://github.com/spacetelescope/mirage/files/3204186/1180_pa_41.xml.zip)

@bd-j

bhilbert4, 2019-06-04:
Sorry for the delayed response. 

Thanks for the suggestion. Parallel imaging with NIRSPec and MIRI was added to the repo not long ago, but only for a subset of the NIRSpec and MIRI observing modes. MIRI MRS should be recognized, but I don't think NIRSpec MOS is. I'll look into it and see if I can add something to cover MOS.

stacchella, 2019-06-05:
It would be great to add something to cover MOS; thank you for looking into it!

bhilbert4, 2021-03-04:
Resolved via #637

closed 2021-03-04 by bhilbert4