spacetelescope/mirage #304: Problems installing MIRaGe OSX environment from file

laurenmarietta, 2019-06-03:
@shanosborne is trying to re-install MIRaGe. Originally, she was trying to just install with `pip install -e .` and ran into a problem because `jwst-background` wasn't specified in the `setup.py` file (which maybe it should, since it's pip-installable?).

I instructed her to try and set up a MIRaGe environment using the `mirage_environment_osx.yml` file instead, but then she got the following error:
```
Collecting package metadata: done
Solving environment: failed
ResolvePackageNotFound: 
  - nbconvert==5.3.1=py_1
  - qt==5.6.2=h9e3eb04_4
```
She was able to set up the environment successfully after I updated the environment file to say just:
```
  - nbconvert==5.3.1
  - qt==5.6.2
```
But we should maybe investigate updating the environment file so others don't run into this problem? Maybe specifying just the version numbers without the build string (e.g. `h9e3eb04_4`) would help.

bhilbert4, 2019-06-06:
Resolved, via #305 and #309

bhilbert4, 2019-06-06:
Oops. #310, rather than #309.

closed 2019-06-06 by bhilbert4