spacetelescope/mirage #306: ValueError in Imaging_simulator_use_examples.ipynb

astrophysaxist, 2019-06-04:
Hi, I just recently forked mirage, and successfully (I believe) installed it on my Linux box. I decided to run one of the example notebooks Imaging_simulator_use_examples.ipynb in a Python 3 notebook. But it seems to fail at the third block of import statements, specifically 

` from mirage.seed_image import catalog_seed_image`

Eventually failing in the geometry `__init__.py`

> /usr/local/anaconda/anaconda3-Linux-x86_64/lib/python3.7/site-packages/photutils/geometry/\_\_init\_\_.py in <module>
      4 """
      5 
----> 6 from .circular_overlap import *       # noqa
      7 from .elliptical_overlap import *     # noqa
      8 from .rectangular_overlap import *    # noqa
\_\_init\_\_.pxd in init photutils.geometry.circular_overlap()
ValueError: numpy.ufunc size changed, may indicate binary incompatibility. Expected 216 from C header, got 192 from PyObject

Any ideas of what could cause this?

Best,
Greg

bhilbert4, 2019-06-04:
Hi Greg,
I haven't seen this error before. My first guess is that maybe there's some problem with using Python 3.7? I've been running Mirage using 3.6 and haven't seen any problems with `photutils` imports. Can you successfully import `geometry` outside of Mirage?

``from photutils import geometry``

Bryan

laurenmarietta, 2019-06-04:
It might also be a problem with the `photutils` version - it should be 0.6. (We're about to update the docs and install instructions to hopefully make package versions for installation clearer!)

astrophysaxist, 2019-06-06:
Hi @bhilbert4 and @laurenmarietta it looks something is definitely up with my photutils. It gives a ValueError when trying to import geometry and even when I just try importing photutils.

laurenmarietta, 2019-06-07:
Hmm, that's no good. Can you share the traceback of the error you're getting?

Also, if you haven't already I might suggest re-installing a new MIRaGe environment, after pulling down the latest changes on the repo. We updated our environment file and installation documentation: https://mirage-data-simulator.readthedocs.io/en/latest/install.html

astrophysaxist, 2019-06-11:
Just updated and started a new mirage environment et viola things work fine until I need the MIRAGE_DATA. I think the problem was using my python 3.7 version of jupyter. This time I used the jupyter of the mirage environment.

bhilbert4, 2019-06-11:
Glad to hear that the environment issue has been cleared up. As for MIRAGE_DATA, I had thought you were inside STScI, in which case the reference data are available. The data aren't currently available to the public. I have a downloader script that is nearly finished which will allow you to download the reference data. That should be available in the relatively near future.

bhilbert4, 2020-01-07:
Just a note to say that the reference files are now public: https://mirage-data-simulator.readthedocs.io/en/latest/reference_files.html

closed 2020-09-30 by bhilbert4