spacetelescope/mirage #319: PUPIL not being correctly populated in wfss simulator

bhilbert4, 2019-06-25:
Pupil value in my NIRCam WFSS simulation (uncal file) is being set to CLEAR when it should be GRISMR/GRISMC.

bhilbert4, 2019-06-25:
Ah, but this was in yaml files that I had edited manually, so I should have changed the pupil value myself in the Mirage inputs.

The underlying problem is that when I ran CAR-35 (prop 1409) through the yaml generator, it failed to recognize the WFSS exposures as WFSS and instead set them to imaging. I need to take a more in-depth look at this, as I have run WFSS proposals through the yaml_generator successfully in the past.

mperrin, 2019-08-07:
This one's probable related to the PUPIL vs PUPIL_MASK keyword. The fix in #342 may also fix this?

bhilbert4, 2020-01-07:
This is because the observations in proposal 1409 are grism exposures using the NIRCam Engineering Imaging APT template. There is more discussion of this in #321. I'll close this issue and keep that one open until the fix is made.

closed 2020-01-07 by bhilbert4