spacetelescope/mirage #322: Difficulties running the new PSF files for NIRISS

KevinVolkSTScI, 2019-07-04:
I have installed the current version of Mirage on witserv3 to test the new PSF libraries with NIRISS.  Various issues have been found in the test case.  A couple of these I was able to fix myself:
(1) simSIgnals-psf_wing_threshold_file is not defined for NIRISS or the FGS the way it is for NIRCam, which causes issues in reading the .yaml file.  Once all_config_files is changed to

        all_config_files = {'nircam': {'Reffiles-subarray_defs': 'NIRCam_subarray_definitions.list',
                                       'Reffiles-flux_cal': 'NIRCam_zeropoints.list',
                                       'Reffiles-crosstalk': 'xtalk20150303g0.errorcut.txt',
                                       'Reffiles-readpattdefs': 'nircam_read_pattern_definitions.list',
                                       'Reffiles-filter_throughput': 'placeholder.txt',
                                       'simSignals-psf_wing_threshold_file': 'nircam_psf_wing_rate_thresholds.txt'},
                            'niriss': {'Reffiles-subarray_defs': 'niriss_subarrays.list',
                                       'Reffiles-flux_cal': 'niriss_zeropoints.list',
                                       'Reffiles-crosstalk': 'niriss_xtalk_zeros.txt',
                                       'Reffiles-readpattdefs': 'niriss_readout_pattern.txt',
                                       'Reffiles-filter_throughput': 'placeholder.txt',
                                       'simSignals-psf_wing_threshold_file': 'niriss_psf_wing_rate_thresholds.txt'},
                            'fgs': {'Reffiles-subarray_defs': 'guider_subarrays.list',
                                    'Reffiles-flux_cal': 'guider_zeropoints.list',
                                    'Reffiles-crosstalk': 'guider_xtalk_zeros.txt',
                                    'Reffiles-readpattdefs': 'guider_readout_pattern.txt',
                                    'Reffiles-filter_throughput': 'placeholder.txt',
                                    'simSignals-psf_wing_threshold_file': 'fgs_psf_wing_rate_thresholds.txt'}}
 
(at line 518 of catalog_seed_image.py) this error goes away.

(2) There is an issue with the long filter wavelength (i.e. F277W) matching to the WebbPSF gridded PSF file because the code pairs these filters with CLEARP (as is the case in the instrument) but WebbPSF records the other element as CLEAR.  Hence none of the gridded PSF files get selected.  To fix this I added a few lines of code in psf_selection.py after line 283:

        # Change CLEARP to CLEAR to match filter wheel filters to allow a match to WebbPSF
        if file_inst.upper() == 'NIRISS' and file_filt in NIRISS_FILTER_WHEEL_FILTERS:
            if file_pupil == 'CLEARP':
                file_pupil = 'CLEAR'

which requires importing NIRISS_FILTER_WHEEL_FILTERS on line 39 of the code

﻿﻿from mirage.utils.constants import NIRISS_PUPIL_WHEEL_FILTERS, NIRISS_FILTER_WHEEL_FILTERS

Having fixed these issues I then have run into a problem in reading the galaxies list file, leading to an error.  I am specifying the galaxy position in pixels, and this somehow leads to an error in catalog_seed_image.py.  The error text is:

Galaxy list input positions assumed to be in units of pixels.
Galaxy list input radii assumed to be in units of arcsec.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/imaging_simulator.py", line 51, in create
    cat.make_seed()
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/seed_image/catalog_seed_image.py", line 192, in make_seed
    self.seedimage, self.seed_segmap = self.addedSignals()
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/seed_image/catalog_seed_image.py", line 1493, in addedSignals
    galaxyCRImage, galaxy_segmap = self.make_galaxy_image(self.params['simSignals']['galaxyListFile'])
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/seed_image/catalog_seed_image.py", line 2956, in make_galaxy_image
    galaxylist = self.filterGalaxyList(glist, pixflag, radflag, magsys, file)
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/seed_image/catalog_seed_image.py", line 2757, in filterGalaxyList
    indexes, galaxylist = self.remove_outside_fov_sources(indexes, galaxylist, pixelflag, 4096)
  File "/home/kvolk/anaconda3/envs/miragetest/lib/python3.6/site-packages/mirage-0.0.0.dev810+g6c825746-py3.6.egg/mirage/seed_image/catalog_seed_image.py", line 1877, in remove_outside_fov_sources
    good = ((catalog_x > minx) & (catalog_x < maxx) & (catalog_y > miny) & (catalog_y < maxy))
TypeError: '>' not supported between instances of 'Column' and 'int'

The galaxy list file has only 1 entry.  The file is

witserv3 testing > more galaxies_test1.list
# 
# vegamag
# position_pixels
# 
x_or_RA y_or_Dec radius ellipticity pos_angle sersic_index niriss_f090w_magnitude 
niriss_f115w_magnitude niriss_f140m_magnitude niriss_f150w_magnitude niriss_f158m_
magnitude niriss_f200w_magnitude niriss_f277w_magnitude niriss_f356w_magnitude nir
iss_f380m_magnitude niriss_f430m_magnitude niriss_f444w_magnitude niriss_f480m_mag
nitude
  1500.0   1500,0   1.0000  0.160   342.87   5.13   17.0000  17.00000  17.00000  1
7.00000  17.00000  17.00000  17.00000  17.00000  17.00000  17.00000  17.00000  17.
00000

The purpose of the test is to see if the galaxy count rate is correct when I made a fix to the generation of the stamp images, around line 2990

            stamp = self.create_galaxy(entry['radius'], entry['ellipticity'], entry['sersic_index'],
                                       xposang*np.pi/180., entry['counts_per_frame_e'])

which it seems should have entry['countrate_e/s'] as the last parameter in the call.  Changing this fixes the galaxy total counts in a previous version of Mirage but I have not tested it yet with the current master version.

bhilbert4, 2019-07-17:
Thanks for reporting these. Do you have a branch that fixes the first two issues you mention? 
As for the galaxy catalog error, it looks like you have a comma in the Dec value for the object. I'm betting this is the problem.

KevinVolkSTScI, 2019-07-17:
I do not have a branch with the fixes.  I was just working with my conda installation code for testing.

KevinVolkSTScI, 2019-07-17:
I see the comma in the test galaxies list file now that you point it out.  My apologies for not seeing it before.  I will try this again with that fixed but I expect it will be the solution.

bhilbert4, 2019-07-18:
I'm working on a branch that includes these fixes, but I'm confused about the CLEAR vs CLEARP fix you made.

Webbpsf's options for pupil values are CLEARP and MASK_NRM. It doesn't know CLEAR. It also always places filters in the FILTER keyword, regardless of which wheel the filter is in. Mirage keeps track of where all elements are actually located, which is why I inserted the lines starting at 274 in psf_selection. These lines will take the filter and pupil values as read in from the webbpsf files, and move them around to match reality. Filters in the pupil wheel will be placed in the pupil variable, and in that case the filter variable will be set to CLEAR.

```
        if file_inst.upper() == 'NIRISS' and file_filt in NIRISS_PUPIL_WHEEL_FILTERS:
            save_filt = copy(file_filt)
            if file_pupil == 'CLEARP':
                file_filt = 'CLEAR'
            else:
                raise ValueError(('Pupil value is something other than '
                                  'CLEARP, but the filter being used is '
                                  'in the pupil wheel.'))
            file_pupil = save_filt
```

But you said you were seeing errors with the long wavelength filters in the filter wheel, which won't run that code block. In that case, the pupil value should remain as CLEARP in order to match mirage's 
expectations, right?

laurenmarietta, 2019-07-18:
(Just an FYI, I think I address some of these problems in #324. Not sure if that is helpful or unhelpful...)

KevinVolkSTScI, 2019-07-18:
What I was seeing is that the gridded PSF files do not have a match to filters that in the pupil wheel which have say F090W + CLEAR as file_pupil and file_filtt in Mirage (which is correct) but which have file_pupil = CLEARP and file_filt = F090W in the WebbPSF header.  Hence for these filters in the pupil wheel to get a match with WebbPSF we need to (a) replace CLEAR [which is correct] with CLEARP [which is wrong] and (b) put the filter name in file_pupil and the other name in file_filter.  This is what the code does, and it did allow the code to find the proper WebbPSF girdded file.

bhilbert4, 2019-07-18:
@laurenmarietta I see some updates for FGS in your branch, but not for NIRISS. Am I missing other changes somewhere? Also sorry for being so slow to review that branch. I was swamped pre-vacation. I'll get to it soon.

So @KevinVolkSTScI, your fix changes the Mirage data to match the webbpsf header info, while my fix does the opposite.  (Although it looks like you are changing the `file_pupil` rather than `pupil` variable.) I chose to adjust the webbpsf data because it's possible to have webbpsf outputs that don't have PUPIL or PUPIL_MASK keywords at all. So I figured it was more robust to check for those and fill in any missing values before matching.

KevinVolkSTScI, 2019-07-18:
My fix may not be optimum.  It appears to allow a gridded PSF mode file to be selected for the short wavelength filters, I did not check whether it affected other things downstream.

laurenmarietta, 2019-07-18:
Ah, whoops, I'm getting PRs confused - I addressed the PSF wings file issue in #303, and I thought I addressed the CLEARP filter thing somewhere but I can't find it now... huh. Sorry about that.

And no worries about the review, I know basically everyone has been busy and/or away!

PR #328 (2019-07-18): Add psf wing files for niriss and fgs
This PR adds references to the PSF wing files necessary for NIRISS and FGS simulations that use the gridded PSF libraries.

@KevinVolkSTScI, I just used this branch and ran NIRISS simulations using a pupil wheel filter and a filter wheel filter. In both cases the correct PSF files were selected. Can you confirm that this is true with your test case? I didn't make any changes to the psf selection code from what is currently in the master branch.

Closes #322
closed 2019-07-18 by KevinVolkSTScI