spacetelescope/mirage #329: Count rates of extended sources

camipacifici, 2019-07-18:
I tested the count rates in a seed image that includes both point sources and extended sources. The count rates for point sources seem fine, while the count rates for extended sources match the counts per frame instead of the counts per second.

@KevinVolkSTScI identify the problem to be in catalog_source_image.py, at the following lines:
           # First create the galaxy
            stamp = self.create_galaxy(entry['radius'], entry['ellipticity'], entry['sersic_index'],
                                       xposang*np.pi/180., entry['counts_per_frame_e'])

where entry['counts_per_frame_e'] should be entry['countrate_e/s'].

camipacifici, 2019-07-26:
@bhilbert4 , @KevinVolkSTScI 
I checked the new count rates and they look much more reasonable.
However, there is a systematic offset between the input counts and the output counts in the seed image, the output counts been fewer. This systematic difference for point sources is of the order of 0.5%, which I would assume is consistent with the fact that the PSFs are not sampled to infinity. For extended sources, the difference is around 4%, which is a little large. Can this also be cause by the finite footprint of the sersic profiles? Maybe they should be larger.
What do you think?

[image]

bhilbert4, 2019-07-26:
The differences in the point sources make sense to me, due to the finite size of the PSF in the library. There was some discussion along these lines in #134. Did you vary the locations of the point sources on the detector? In past testing we saw the total signal in the PSF vary with position.

As for the extended sources, are you referring to galaxies? I would expect some difference there, but the amount of difference should be highly dependent on the exact values in the parameters (sersic index, radius, eccentricity).  In order to prevent mirage from slowing down too much, it does crop the galaxy stamp image, although missing 4% seems a little extreme.

I'll look into the code on Monday to check the cropping.

camipacifici, 2019-07-27:
Yes, I meant galaxies.
I can try a more systematic combination of the various parameters and positions on the detector. I will do it on Monday and send you the images and the count comparison.

bhilbert4, 2019-08-09:
I've tracked down this difference in brightness to the step where the stamp image of the galaxy is convolved with the PSF. The total signal of the PSF is less than 1, since webbpsf normalizes the PSF at the entrance pupil, and we're not dealing with an infinite aperture. The latter is the dominant reason here. In my test cases, I was looking at NIRCam and NIRISS imaging modes, and the integrated PSF signal was 0.976 for NIRCam and 0.955 for NIRISS. This was for PSF stamp images with sizes of 65x65 pixels for NIRCam and 23x23 pixels for NIRISS. These stamp sizes are based on the sizes of the PSFs in the gridded library files, which were chosen by searching for a consistent encircled energy level.

When adding point sources to the simulation, the PSF "cores" above have the wings out to some user-specified distance added onto them. But when convolving with galaxies and extended sources, the wings are not added to the PSF cores. I did it this way to save computation time, and hadn't thought about the loss of signal.

So for the purposes of convolving with galaxies and extended images, it seems like we should we add the wings to the PSF first. I wondered for a second whether we could just renormalize the PSF core to a total integrated signal of 1.0, but this would be inconsistent with the way point sources are treated.

@mperrin does that make sense to you?

bhilbert4, 2019-08-09:
I should also mention that I tested using galaxies on a few different locations across the detectors, as well as for a wide range of sersic indices, but the differences from these factors were trivial compared to the effect from the integrated signal of the PSF.

mperrin, 2019-08-09:
It makes sense to me, yes. My sense is it might be sufficient to normalize the PSFs to 1 as you suggested. I see the consistency argument for using the PSF wings instead but I wonder about the performance cost for that. If it’s efficient enough to live with then sure, seems like a reasonably way to proceed. 

For extended sources I suspect the field dependence of PSFs is a negligible effect, yes.

bhilbert4, 2019-08-10:
Looks like adding the wings to the PSFs is roughly doubling the computation time. I tested a scene with 200 galaxies and generating the seed image goes from taking about 5 seconds to about 11 seconds.

PR #347 (2019-08-14): Normalize PSF to convolve with galaxy and extended image stamps
This PR fixes a bug that was causing galaxy and extended images to have signal levels that were too low. This was due to the PSF that is convolved with the galaxy/extended stamp images. Since we are using just the "core" of the PSF for the convolution, and the full PSF is normalized to 1.0 at the telescope entrance pupil for an infinite aperture, the total signal in our "core" is several percent below 1.0. This was then causing the total signal in the convolved galaxy/extended image to be several percent lower than expected. 

We  [... cut]
closed 2019-08-14 by bhilbert4