spacetelescope/mirage #331: Allow for more intuitive filter/pupil listings in yaml files

bhilbert4, 2019-07-19:
Mirage currently expects filter and pupil values in the input yaml files that agree with the elements in the real filter and pupil wheels. For example, the NIRISS F090W filter is in the pupil wheel. In order to create F090W data, 'F090W' must be listed under the 'pupil' entry in the yaml file. However, the F444W filter is in the filter wheel, meaning 'F444W' must be in the 'filter' entry of the yaml file.

This requires users to know which filters are present in each wheel for a particular instrument. Mirage should be updated so that a user can put any filter name in the 'filter' entry of the yaml file and Mirage will know what to do.

Currently, the yaml generator creates yaml files that reflect reality. Pupil wheel filters are listed in the 'pupil' entry, etc.

I think the easiest solution would be to create a function that examines the filter and pupil values as read from the yaml file and if necessary, translates them to reflect the true locations of the elements. In this way the yaml generator would not have to be changed and Mirage will still operate based on the actual locations of the elements, but users manually creating yaml files would still have the convenience of being able to place any filter name in the 'filter' entry. Note that with this convenience can come a little confusion.  A user could, for example, put F200W in the 'filter' entry and 'NRM' in the pupil entry. In reality these two elements are both in the pupil wheel and cannot be used together. In that case, the translating function should probably throw an error stating that the filter/pupil combination is not valid.

@KevinVolkSTScI, what do you think?

KevinVolkSTScI, 2019-07-19:
Yes this all sounds good, I can try to write such a function.  I should be able to find where the yaml file is read and make something at that point to examine the FILTER and PUPIL elements when the instrument is NIRISS.

bhilbert4, 2019-07-19:
I can also envision a similar function for NIRCam, which deals with the narrow band filters in the pupil wheel. In that case I can see allowing a user to enter one of the pupil wheel filters in the 'filter' entry, and have the function choose the appropriate matching filter wheel filter, and then placing the two filters in the appropriate 'filter' and 'pupil' entries.

For example, a user can enter 'F405N' in the 'filter' entry, and the function would translate this into a 'pupil' value of F405N and a 'filter' value of F444W.