spacetelescope/mirage #337: error when in yaml_generator.SimInput

mgennaro, 2019-07-31:
I was running the quickstart_for_hack_day notebook, in particular the following commands:

`yam = yaml_generator.SimInput()
yam.input_xml = './apt_files/example_for_hack_day.xml'
yam.pointing_file = './apt_files/example_for_hack_day.pointing'
yam.output_dir = './yaml_files/'
yam.simdata_output_dir = './simulated_data/'
yam.observation_table = 'observation_list.yaml'
yam.use_JWST_pipeline = True
yam.use_linearized_darks = False
yam.datatype = 'raw'
yam.reffile_setup()
yam.create_inputs()`

The first command generates an error:
`---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-7-f884bde40b63> in <module>
      1 # Create a series of data simulator input yaml files
      2 # from APT files
----> 3 yam = yaml_generator.SimInput()
      4 yam.input_xml = './apt_files/example_for_hack_day.xml'
      5 yam.pointing_file = './apt_files/example_for_hack_day.pointing'

~/anaconda3/envs/mirage/lib/python3.6/site-packages/mirage-0.0.0.dev861+g08a92050-py3.6.egg/mirage/yaml/yaml_generator.py in __init__(self, input_xml, pointing_file, datatype, use_JWST_pipeline, catalogs, observation_list_file, verbose, output_dir, simdata_output_dir, parameter_defaults, offline)
    174 
    175         self.set_global_definitions()
--> 176         self.path_defs()
    177 
    178         if (input_xml is not None) and (catalogs is not None):

~/anaconda3/envs/mirage/lib/python3.6/site-packages/mirage-0.0.0.dev861+g08a92050-py3.6.egg/mirage/yaml/yaml_generator.py in path_defs(self)
   1010     def path_defs(self):
   1011         """Expand input files to have full paths"""
-> 1012         self.input_xml = os.path.abspath(os.path.expandvars(self.input_xml))
   1013         self.pointing_file = os.path.abspath(os.path.expandvars(self.pointing_file))
   1014         self.output_dir = os.path.abspath(os.path.expandvars(self.output_dir))

~/anaconda3/envs/mirage/lib/python3.6/posixpath.py in expandvars(path)
    286     """Expand shell variables of form $var and ${var}.  Unknown variables
    287     are left unchanged."""
--> 288     path = os.fspath(path)
    289     global _varprog, _varprogb
    290     if isinstance(path, bytes):

TypeError: expected str, bytes or os.PathLike object, not NoneType
`

This is because the user is allowed to creat an instance of SimInput w/o passing any files, yet the __init__ code fails when calling the path_defs method, which does expect xml and pointing files  to have been passed (the two have defaults equal to None). I believe the solution is to force the user to pass those files at instantiation, unless there are use cases for this class to be used w/o those inputs

bhilbert4, 2019-07-31:
Thanks. I'll make an update. If I recall correctly, we allowed the case of instantiating the class without providing filenames in order to make testing easier. But anyway, I'll adjust the code to avoid this error.

mgennaro, 2019-07-31:
Ok, I made it through that step...but now there's a new error, again possibly related to not forcing the user to specify some parameters. This is my new code:

```
yam = yaml_generator.SimInput(input_xml='./apt_files/example_for_hack_day.xml',
                              pointing_file = './apt_files/example_for_hack_day.pointing',
                              output_dir = './yaml_files/',
                              simdata_output_dir = './simulated_data/',
                              observation_list_file = 'observation_list.yaml')

yam.use_JWST_pipeline = True
yam.use_linearized_darks = False
yam.datatype = 'raw'
yam.reffile_setup()
yam.create_inputs()
```

and here's the error:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-9-0ac70b307ca0> in <module>
     15 yam.datatype = 'raw'
     16 yam.reffile_setup()
---> 17 yam.create_inputs()

~/anaconda3/envs/mirage/lib/python3.6/site-packages/mirage-0.0.0.dev861+g08a92050-py3.6.egg/mirage/yaml/yaml_generator.py in create_inputs(self)
    308             # apt.pointing_file = self.pointing_file
    309             apt.observation_list_file = self.observation_list_file
--> 310             apt.apt_xml_dict = self.apt_xml_dict
    311 
    312             apt.output_dir = self.output_dir

AttributeError: 'SimInput' object has no attribute 'apt_xml_dict'
```

which in turn, I believe is related to the code not entering this if clause within the `__init__` method:

```
        if (input_xml is not None) and (catalogs is not None):
            if self.observation_list_file is None:
                self.observation_list_file = os.path.join(self.output_dir, 'observation_list.yaml')
            self.apt_xml_dict = get_observation_dict(self.input_xml, self.observation_list_file, self.catalogs,
                                                     verbose=self.verbose, parameter_defaults=parameter_defaults)
```

because `catalog` in now set to the default `None`  and thus that `self.apt_xml_dict` attribute is never set

bhilbert4, 2019-07-31:
I'll modify that as well, probably by changing the default to be 'none' from None.
Also, be careful providing the observation_list_file in the instantiation. If you do, make sure it's been created from a recent run of the code because the format did change a while back. Safest is not to provide it at all and a new one will be generated.

bhilbert4, 2019-07-31:
Also, self.reffile_setup is now run at instantiation, so you shouldn't have to call that yourself.

mgennaro, 2019-07-31:
If I remove the `observation_list_file` I get this:

```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
<ipython-input-8-0d1059025b6f> in <module>
     10 yam.datatype = 'raw'
     11 #yam.reffile_setup()
---> 12 yam.create_inputs()

~/anaconda3/envs/mirage/lib/python3.6/site-packages/mirage-0.0.0.dev861+g08a92050-py3.6.egg/mirage/yaml/yaml_generator.py in create_inputs(self)
    328 
    329         else:
--> 330             raise FileNotFoundError(("WARNING. You must include either an ascii table file of observations"
    331                                      " or xml and pointing files from APT plus the observation list file."
    332                                      "Aborting."))

FileNotFoundError: WARNING. You must include either an ascii table file of observations or xml and pointing files from APT plus the observation list file.Aborting.
```

bhilbert4, 2019-07-31:
Looks like you are running an old version of Mirage. Did you install when you fetched the code just now? Version number should be 1.0.x.xxx

PR #338 (2019-08-15): Update defaults for catalogs, and allow no xml or pointing file to be passed
Closes #337 

Small updates to `yaml_generator` that allows a user to instantiate the class without providing xml or pointing files. This is potentially useful for testing.

Also updated the default value for `catalogs` in the case where they are not provided. This way people can create yaml files without providing the names of input catalogs. All catalogs are set to 'none'.

@mgennaro do you want to try using this branch to see if it works for you? I tried running the yaml_generator cell in the example imaging notebo [... cut]
closed 2019-08-15 by bhilbert4